mavlink/qgroundcontrol #1674: CP210x modem not recognised by QGC

jayceelock, 2015-06-22:
Hi,

I tried connecting a set of USB-to-UART telemetry modems to the Pixhawk and laptop for logging purposes. For some reason though, QGC doesn't seem to pick the device up. `dmesg` shows that it gets registered under ttyUSB0 with the cp210x driver and is listed when I run `lsmod`. Running `modprobe cp210x` doesn't produce any errors either.

I tested it with Mint 17.1 as well as Ubuntu 14.04. According to [this](https://www.pololu.com/docs/0j7/all) site, the cp210x was included in Ubuntu 8.04 already. Is there something I'm missing, or does QGC not support these modems under Linux?

Funnily enough, there's a working driver available for Windows and QGC seems to recognising the modem there just fine.

DonLakeFlyer, 2015-06-22:
> For some reason though, QGC doesn't seem to pick the device up.

Which means what? Exactly how to you attempt to connect to it?

jayceelock, 2015-06-22:
Sorry, let me rephrase. QGC picks up the modem connected to the laptop on
ttyUSB0, but it doesn't connect to the partner modem on the Pixhawk

jayceelock, 2015-06-22:
Nevermind, I'm an idiot. I just realised I plugged the modem into the TELEM2 port and not the TELEM1 port...

It's still interesting that it works with an older version of QCG in Windows. Is this due to an update to QCG or to the PX4 firmware?

DonLakeFlyer, 2015-06-22:
I doubt this has anything to do with QGC versions.

jayceelock, 2015-06-24:
I'm still having trouble with QGC not detecting my modem. It works fine with the Stable V2.2 branch, but in all versions after that, QGC does not automatically list it as a device and manually adding it does not seem to fix it either. All versions still work with my 3DR 900MHz modem.

DonLakeFlyer, 2015-06-24:
Only 3DR Radio's will be automatically detected. All other radio connections must be added manually. Use Add Link or Preferences.

closed 2015-06-28 by DonLakeFlyer