mavlink/qgroundcontrol #1682: Displaying simple 3D model of vehicle

boosfelm, 2015-06-26:
We discussed with @LorenzMeier and @tumbili that a widget displaying a simple 3D model of an airframe (maybe only quad, vtol, fixed-wing) + current attitude and attitude setpoint as coordinate frames would be super helpful to assess angle tracking at a first glance.

This would also partially solve the problem with displaying gimbal lock in the flight display for VTOL atm, because we could generate the coordinate frames from quaternions / dcm.
Waht do you think on that, where would that fit?

DonLakeFlyer, 2015-06-26:
Don't understand how that differs from the HUD? What value does the 3d model add?

DonLakeFlyer, 2015-06-27:
I think I should have dais Primary Flight display instead of HUD.

boosfelm, 2015-06-29:
Hi Don
Sorry but now I don't get you completely, what does "dais" mean? For me the idea of the 3D model makes sense, because it displays all your three errors and thus you have a first intuition whether your angle tracking works well. It is quiet tough to include this in a HUD a suppose. 
@tumbili did a very basic visualization for himself.
[image]

dogmaphobic, 2015-06-29:
>I think I should have dais Primary Flight display instead of HUD.

Ultimate dyslexia? :)

I think he meant:

"I think I should have *said* Primary Flight display instead of HUD."

DonLakeFlyer, 2015-06-29:
Ok. I get it now.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier