mavlink/qgroundcontrol #1689: Mavlink FTP download of files fails

boosfelm, 2015-06-30:
With latest stable qgc (2.5.2) I cannot download logging data via mavlink ftp. In the onboard files widget I try to list files and it stops with an timeout error - any thoughts on that?
[image]

DonLakeFlyer, 2015-06-30:
This is over USB? FTP doesn’t work over radio.

boosfelm, 2015-07-01:
thanks for the quick reply. First trial was indeed via telemetry, but also via FTDI I can only download very small files (preflight / postflight) but no log file. It seems to have difficulties handling package loss.
[image]

LorenzMeier, 2015-07-01:
@boosfelm I presume it works with a different QGC version? If yes, with which one?

LorenzMeier, 2015-07-01:
@DonLakeFlyer I'm not aware about recent FTP changes on the Firmware side.

DonLakeFlyer, 2015-07-01:
Nothing has changed here that recently. 

Now you say it doesn’t work over FTDI. Does it work over USB?

boosfelm, 2015-07-05:
Hi @DonLakeFlyer and @LorenzMeier 
Thanks for your reply. Today I have tried via usb and it works fine, although rather slow. I went for 921600 baud rate and got down to 11MB in 10 min.
I really like the idea of it, but I believe it is still somewhat cumbersome and thus only few people in the community use it. Connecting via usb is probably similar to removing the sd_card, which will be always much faster. @LorenzMeier Really looking forward to LTE integration ;)

[image]

boosfelm, 2015-07-05:
it seems to me that you can only start mavlink on an usb port when usb connected. Thus the pixhawk has to be powered externally, connect usb cable and start mavlink via nsh and connect via qgc. Else the following error message occurs.
[image]

DonLakeFlyer, 2015-07-22:
Working as well as it can over USB

closed 2015-07-22 by DonLakeFlyer