mavlink/qgroundcontrol #1704: FactTextField is showing way too many decimal values

dogmaphobic, 2015-07-04:
Something got messed up with FactTextField, which is now showing way too many values after the decimal point, which in turn is messing up the formatting of the text field.

[image]

dogmaphobic, 2015-07-04:
Hmm.. I didn't realized I was using Qt 5.5.0 for this build. This issue is only present in Qt 5.5.0. Nonetheless, we need to find out what is causing it.

DonLakeFlyer, 2015-07-04:
Something changed with QVariant::toString on float value which is what valueString calls.

LorenzMeier, 2015-09-01:
This is an official Qt 5.5 bug, scheduled to be fixed until mid September.

closed 2015-10-25 by DonLakeFlyer