mavlink/qgroundcontrol #1708: QGroundControl-Installer-Win32-Stable-V2.5.2

prprdm, 2015-07-08:
Problem signature:
  Problem Event Name:	APPCRASH
  Application Name:	qgroundcontrol.exe
  Application Version:	0.0.0.0
  Application Timestamp:	558f40aa
  Fault Module Name:	ig4icd32.dll
  Fault Module Version:	7.15.10.1502
  Fault Module Timestamp:	4851ca59
  Exception Code:	c0000005
  Exception Offset:	0004d2e6
  OS Version:	6.0.6002.2.2.0.768.3
  Locale ID:	4105
  Additional Information 1:	fd00
  Additional Information 2:	ea6f5fe8924aaa756324d57f87834160
  Additional Information 3:	fd00
  Additional Information 4:	ea6f5fe8924aaa756324d57f87834160

Read our privacy statement:
  http://go.microsoft.com/fwlink/?linkid=50163&clcid=0x0409

DonLakeFlyer, 2015-07-08:
Your OpenGL driver is crashing. Update your video drives if possible. If they are already up to date then you will need to wait until we switch to Qt 5.5 to be able to run QGC on this machine. That is in the works, but still doesn’t have an exact date.

prprdm, 2015-07-08:
THANKS! I did update my video drivers... The weird thing is it crashes on both my laptop's...appreciate the reply

https://github.com/mavlink/qgroundcontrol/issues/1708#issuecomment-119387834

DonLakeFlyer, 2015-07-08:
How old is the laptop and what model?

prprdm, 2015-07-08:
Definitely older. Toshiba satellite L300 if I'm not mistaken

Sent from my Samsung device

closed 2015-07-22 by LorenzMeier