mavlink/qgroundcontrol #1713: parameter change in custom widget

flyingk, 2015-07-09:
When trying to change a parameter on my own autpilot in the custom command widget by pressing enter or clicking the following error happens
qrc:///qml/QGroundControl/FactControls/FactTextField.qml:21: ReferenceError: qgcView is not defined

If I comment out the reference for qgcView in that file it works. Maybe this is a PX4 specific thing?

DonLakeFlyer, 2015-07-09:
Can you provide your custom Qml? Also please send problems to the mailing list first before putting in a GitHub issue.

DonLakeFlyer, 2015-07-09:
Never mind with sending the Qml. There's a problem with using a FactPanel as the outer control. Need to figure out how to fix it.

DonLakeFlyer, 2015-07-09:
Add this to your FactPanel as a work around until I figure out real fix:

property var qgcView: null // Temporary hack for broken QGC parameter validation implementation

Example updated: http://www.qgroundcontrol.org/custom_command_qml_widgets

DonLakeFlyer, 2015-07-22:
@flyingk Is this working for you now?

flyingk, 2015-07-22:
Yes it works fine now. Thanks for the fix.

Kai

closed 2015-07-22 by DonLakeFlyer