mavlink/qgroundcontrol #1714: Parameter Editor, mavlink inspector, etc won't show up after connection

shengwen1997, 2015-07-11:
Environment: Elementary OS freya(based on Ubuntu 14.04)
Version: 2.5 stable and newest daily build
Control board: self-made

This is a pretty annoying problem for me.
Sometime when I connect my quadcopter  to the ground station,
the parameter editor, mavlink inspector and some component won't show up.

Just like this picture. Although the status bar show me the connection is well, the parameter editor just
said: "no vehicle connected", also, the attitude is still plot perfectly.
[image]

LorenzMeier, 2015-07-22:
If you boot the system incompletely you may end up with this state. @DonLakeFlyer Can you think of a way to make the internal state more transparent so its easier to debug for flight code developers? I fixed the boot timing for good on beta and stable now.

DonLakeFlyer, 2015-07-23:
Better reporting now complete.

closed 2015-07-23 by DonLakeFlyer