mavlink/qgroundcontrol #1718: Firmware upgrade sync error

AndreasAntener, 2015-07-15:
With master from just now, I get the following error on the firmware upgrade:
```
Plug in your device via USB to start firmware upgrade
Found device: Pixhawk
Error: Sync: Send Command: Invalid sync response: 0x21 0x20
Upgrade cancelled
```

LorenzMeier, 2015-07-20:
@DonLakeFlyer I get the same error.

DonLakeFlyer, 2015-07-20:
Why would the bootloader be sending back 21/20? That's PROTO_GET_SYNC/PROTO_EOC! Are you running normal bootloaders?

AndreasAntener, 2015-07-20:
Nothing special there.

DonLakeFlyer, 2015-07-21:
100% repro? Everything ok when I try.

DonLakeFlyer, 2015-07-21:
@LorenzMeier I can't even see how the bootloader could have sent these bytes back. There is no code that does that in bootloader. You may need to help out here and step through it in the debugger to see what is going on since I can't repro on any of my boards.

AndreasAntener, 2015-07-21:
I just tried with a fresh Pixhawk from 3DR and it works there. So my other one must have an older bootloader?

DonLakeFlyer, 2015-07-21:
Possibly, but it should still work. You have 100% failure on your other board? Do you build firmware source? Wondering is px4-fmuv2_default upload works ok, but QGC fails?

LorenzMeier, 2015-07-21:
Default definitely works in that state

DonLakeFlyer, 2015-07-21:
@LorenzMeier Ok. You're going to have to help with debugging here. Without a repro on my side I can't do much. I look at the bootloader and the qgc upgrade code to look for possibilities but couldn't find anything that jumped out from code inspection.

DonLakeFlyer, 2015-07-22:
Can you run QGC with the --full-logging command line option and email me the log? May get lucky. But still think I won't be able to fix this on my end since I can't repro.

AndreasAntener, 2015-07-22:
Should I find a log file somewhere or just the console out?

I just noticed a difference between the 2 Pixhawks: the "old one" doesn't stay at the bootloader but boots up normally, then QGC reports the sync error.

LorenzMeier, 2015-07-22:
Then it doesn't know about the reboot into continuous bootload mode (reboot -b) flag.
@DonLakeFlyer This suggests that the timing broke and the QGC code is now too slow to catch the boot loader.

DonLakeFlyer, 2015-07-22:
@AndreasAntener Can you boot the board that doesn't flash by just plugging into USB and watch the leds on it. The PWR leds under the FMU label should turn on and the B/E led under the FMU label should flash for about 3 seconds and the go out. is it doing that?

DonLakeFlyer, 2015-07-22:
Also should have said that --full-logging is console output

DonLakeFlyer, 2015-07-22:
Ok, I think I found it through code inspection. There's a wait in there for the Radio bootloader which is putting things right on the edge of missing the bootloader. Need to figure out how to fix, then I need to g through all the radio/pixhawk/flow testing again.

DonLakeFlyer, 2015-07-22:
Please try latest daily build: http://www.qgroundcontrol.org/downloads for the fix, or build your own from master. Let me know.

AndreasAntener, 2015-07-22:
Cool, will try tomorrow (don't have that Pixhawk with me)

DonLakeFlyer, 2015-07-23:
Any news? Working on new stable with this fix. Need to know if it really works.

AndreasAntener, 2015-07-24:
Sorry for the late reply. Didn't work for me:
```
FirmwareUpgradeLog: PX4FirmwareUpgradeThreadController::findBoard
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: Found PX4 FMU V2
FirmwareUpgradeLog: _findBootloader
FirmwareUpgradeLog: Bootloader error: "Sync: Send Command: Invalid sync response: 0x21 0x20"
FirmwareUpgradeLog: "Pixhawk"
FirmwareUpgradeLog: PX4FirmwareUpgradeThreadController::cancel`
```

DonLakeFlyer, 2015-07-24:
Are you setup to get serial output from your board on telem 4/5? If so could you watch exactly when you get the Bootloader error and see if it happens before you start getting firmware output from the serial console? Other way is to watch the lights as a decribed a few posts pack. The light sequence for the board in bootloader mode is the B/E light flashing on the FMU side. When that  light goes out the bootloader is not longer running. Trying to determine if somehow upgrade is missing the bootloader.

AndreasAntener, 2015-07-27:
The error happens after bootloader mode (so after I get output on serial and after the FMU B/E leds stopped flashing).

Also, QGC output shows me these messages after bootloader mode:
```
FirmwareUpgradeLog: Found PX4 FMU V2
FirmwareUpgradeLog: _findBootloader
```
Before it's still repeating `_findBoardOnce`

AndreasAntener, 2015-07-27:
If I plug in the new Pixhawk, I get these messages instantly:
```
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: Found PX4 FMU V2
FirmwareUpgradeLog: _findBootloader
FirmwareUpgradeLog: "Pixhawk"
FirmwareUpgradeLog: Found bootloader
```

AndreasAntener, 2015-07-27:
So it seems that the serial port doesn't show up until it is past bootloader mode.

```
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: Serial Port --------------
FirmwareUpgradeLog: 	port name: "Bluetooth-Incoming-Port"
FirmwareUpgradeLog: 	description: "incoming port - Bluetooth-Incoming-Port"
FirmwareUpgradeLog: 	system location: "/dev/cu.Bluetooth-Incoming-Port"
FirmwareUpgradeLog: 	vendor ID: 0
FirmwareUpgradeLog: 	product ID: 0
FirmwareUpgradeLog: Serial Port --------------
FirmwareUpgradeLog: 	port name: "Bluetooth-Modem"
FirmwareUpgradeLog: 	description: "Bluetooth-Modem"
FirmwareUpgradeLog: 	system location: "/dev/cu.Bluetooth-Modem"
FirmwareUpgradeLog: 	vendor ID: 0
FirmwareUpgradeLog: 	product ID: 0
FirmwareUpgradeLog: Serial Port --------------
FirmwareUpgradeLog: 	port name: "usbserial-A601ZKB2"
FirmwareUpgradeLog: 	description: "FT232R USB UART"
FirmwareUpgradeLog: 	system location: "/dev/cu.usbserial-A601ZKB2"
FirmwareUpgradeLog: 	vendor ID: 1027
FirmwareUpgradeLog: 	product ID: 24577
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: Serial Port --------------
FirmwareUpgradeLog: 	port name: "Bluetooth-Incoming-Port"
FirmwareUpgradeLog: 	description: "incoming port - Bluetooth-Incoming-Port"
FirmwareUpgradeLog: 	system location: "/dev/cu.Bluetooth-Incoming-Port"
FirmwareUpgradeLog: 	vendor ID: 0
FirmwareUpgradeLog: 	product ID: 0
FirmwareUpgradeLog: Serial Port --------------
FirmwareUpgradeLog: 	port name: "Bluetooth-Modem"
FirmwareUpgradeLog: 	description: "Bluetooth-Modem"
FirmwareUpgradeLog: 	system location: "/dev/cu.Bluetooth-Modem"
FirmwareUpgradeLog: 	vendor ID: 0
FirmwareUpgradeLog: 	product ID: 0
FirmwareUpgradeLog: Serial Port --------------
FirmwareUpgradeLog: 	port name: "usbserial-A601ZKB2"
FirmwareUpgradeLog: 	description: "FT232R USB UART"
FirmwareUpgradeLog: 	system location: "/dev/cu.usbserial-A601ZKB2"
FirmwareUpgradeLog: 	vendor ID: 1027
FirmwareUpgradeLog: 	product ID: 24577
FirmwareUpgradeLog: _findBoardOnce
FirmwareUpgradeLog: Serial Port --------------
FirmwareUpgradeLog: 	port name: "Bluetooth-Incoming-Port"
FirmwareUpgradeLog: 	description: "incoming port - Bluetooth-Incoming-Port"
FirmwareUpgradeLog: 	system location: "/dev/cu.Bluetooth-Incoming-Port"
FirmwareUpgradeLog: 	vendor ID: 0
FirmwareUpgradeLog: 	product ID: 0
FirmwareUpgradeLog: Serial Port --------------
FirmwareUpgradeLog: 	port name: "Bluetooth-Modem"
FirmwareUpgradeLog: 	description: "Bluetooth-Modem"
FirmwareUpgradeLog: 	system location: "/dev/cu.Bluetooth-Modem"
FirmwareUpgradeLog: 	vendor ID: 0
FirmwareUpgradeLog: 	product ID: 0
FirmwareUpgradeLog: Serial Port --------------
FirmwareUpgradeLog: 	port name: "usbserial-A601ZKB2"
FirmwareUpgradeLog: 	description: "FT232R USB UART"
FirmwareUpgradeLog: 	system location: "/dev/cu.usbserial-A601ZKB2"
FirmwareUpgradeLog: 	vendor ID: 1027
FirmwareUpgradeLog: 	product ID: 24577
FirmwareUpgradeLog: Serial Port --------------
FirmwareUpgradeLog: 	port name: "usbmodem1"
FirmwareUpgradeLog: 	description: "PX4 FMU v2.x"
FirmwareUpgradeLog: 	system location: "/dev/cu.usbmodem1"
FirmwareUpgradeLog: 	vendor ID: 9900
FirmwareUpgradeLog: 	product ID: 17
FirmwareUpgradeLog: Found PX4 FMU V2
FirmwareUpgradeLog: _findBootloader
FirmwareUpgradeLog: Bootloader error: "Sync: Send Command: Invalid sync response: 0x21 0x20"
FirmwareUpgradeLog: "Pixhawk"
FirmwareUpgradeLog: PX4FirmwareUpgradeThreadController::cancel
```

DonLakeFlyer, 2015-07-27:
Yup. it's never seeing the bootloader. Try the latest daily build. I've sped up the board find loop to hopefully not miss the bootloader. If that doesn't work, I'm not sure what to do. You can flash this board with the firmware python uploader right?

AndreasAntener, 2015-07-27:
Looks like I had your change already in when I tested today :/
Upload via make upload works.

DonLakeFlyer, 2015-07-27:
Crap! When you are using the firmware uploader does it catch the bootloader on first try or is it having to send the reboot command? Try this:
- unplug board
- make px4fmu-v2_default upload
- plug in board

Does is start flash right away on bootloader according to leds? Or does it have to send the reboot command. I think in that case it shows some more text above having to wait.

AndreasAntener, 2015-07-27:
It starts right away. I just compared the 2 methods used: The python uploader just tries certain serial port names depending on the OS, which is hacky. The port name pops up fast, also in QGC. But what I just found out is that the product identifier changes. It has 22 when it's still in bootloader mode, but 17 afterwards.

AndreasAntener, 2015-07-27:
Yeah, must be it, just add 22 to the list ;)
The new Pixhawk turns up with 17 right away. No clue why the difference (especially why it changes after bootloader mode)

DonLakeFlyer, 2015-07-27:
Yup, that's it. Great find. Thanks a ton.

DonLakeFlyer, 2015-07-27:
Master is updated. New Daily builds in about 20 minutes. Let me know.

AndreasAntener, 2015-07-28:
Works, thanks :)

closed 2015-07-28 by AndreasAntener

DonLakeFlyer, 2015-07-28:
Awesome. Thanks for sticking with it to help us figure this one out. Good find.

bradsedito, 2015-10-03:
I'm still having this issue and it's literally driving me nuts.... anyone feel like helping me fix this? Madness!!

paralin, 2019-06-29:
I'm having this problem with an old Pixhawk, any fix?

pgautam, 2021-08-04:
I am also facing the same issue, 

I am using latest QGroundControl on WIndows10 created on 2021-05-06
[image]

When trying to upload Firmware, "Upgrade cancelled", I am using NXP's PX4 FMUK66 v3.x

[image]

Any help to resolve this issue.

dgorissen, 2025-03-02:
Just bumped into this as Im ressurecting an old drone. Connecting directly to pixhawk is fine. Via radios is not working. Trying to upgrade firmware:

```
 QGroundControl can upgrade the firmware on Pixhawk devices, SiK Radios and PX4 Flow Smart Cameras.
All QGroundControl connections to vehicles must be disconnected prior to firmware upgrade.
Please unplug your Pixhawk and/or Radio from USB.
Plug in your device via USB to start firmware upgrade.
Found device: SiK Radio
Connected to bootloader:
Version: 0
Board ID: 78
Flash size: 0
Downloading firmware...
From: http://px4-travis.s3.amazonaws.com/SiK/stable/radio~hm_trp.ihx
Download complete
Error: Sync: Send Command: Invalid sync response: 0x10 0x12
Upgrade cancelled
------------------------------------------
If upgrade failed, make sure to connect directly to a powered USB port on your computer, not through a USB hub. Also make sure you are only powered via USB not battery.
```

With console output

```
~  ./QGroundControl.AppImage --logging:FirmwareUpgradeLog                                                                                              ✔ │ at 09:54:55 
Settings location "/home/dgorissen/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\nFirmwareUpgradeLog.debug=true\nGStreamerAPILog.debug=true\nqt.qml.connections=false"
System reported locale: QLocale(English, Latin, United Kingdom) ; Name "en_GB" ; Preffered (used in maps):  "en-GB"
LocalizationLog: Qt lib localization for "en_GB" is not present
LocalizationLog: Error loading source localization for "en_GB"
LocalizationLog: Error loading json localization for "en_GB"
serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT.
VideoReceiverLog: Stop called on empty URI
VideoReceiverLog: Stop called on empty URI
MAVLinkLogManagerLog: MAVLink logs directory: "/home/dgorissen/Documents/QGroundControl/Logs"
qt.network.ssl: QSslSocket: cannot resolve EVP_PKEY_base_id
qt.network.ssl: QSslSocket: cannot resolve SSL_get_peer_certificate
Ignoring ssl certificates due to OpenSSL version mismatch
Map Cache in: "/home/dgorissen/.cache/QGCMapCache300" / "qgcMapCache.db"
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
setCurrentPlanViewSeqNum
setCurrentPlanViewSeqNum
qt.network.ssl: QSslSocket: cannot call unresolved function EVP_PKEY_base_id
"v4.4.3"
Ignoring ssl certificates due to OpenSSL version mismatch
Ignoring ssl certificates due to OpenSSL version mismatch
FirmwareUpgradeLog: PX4FirmwareUpgradeThreadController::findBoard
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
qt.network.ssl: QSslSocket: cannot call unresolved function EVP_PKEY_base_id
"FTDI" "FT230X Basic UART"
FirmwareUpgradeLog: 1 "SiK Radio"
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get_peer_certificate
qt.network.ssl: QSslSocket: cannot call unresolved function EVP_PKEY_base_id
Ignoring ssl certificates due to OpenSSL version mismatch
FirmwareUpgradeLog: Found px4 stable version "v1.15.4"
FirmwareUpgradeLog: Found px4 beta version "v1.16.0-alpha2"
FirmwareUpgradeLog: _ardupilotManifestDownloadFinished "https://firmware.ardupilot.org/manifest.json.gz" "/tmp/manifest.json.gz"
FirmwareUpgradeLog: Board gone
FirmwareUpgradeLog: open: "ttyUSB0"
"FTDI" "FT230X Basic UART"
FirmwareUpgradeLog: 1 "SiK Radio"
FirmwareUpgradeLog: Radio in normal mode
FirmwareUpgradeLog: FirmwareUpgradeController::flash stackType:firmwareType:vehicleType FirmwareUpgradeController::SiKRadio FirmwareUpgradeController::StableFirmware FirmwareUpgradeController::DefaultVehicleFirmware
Ignoring ssl certificates due to OpenSSL version mismatch
FirmwareUpgradeLog: "_ihxLoad - EOF"
FirmwareUpgradeLog: PX4FirmwareUpgradeThreadWorker::_flash
FirmwareUpgradeLog: PX4FirmwareUpgradeThreadController::cancel
FirmwareUpgradeLog: reboot ATZ

```

DonLakeFlyer, 2025-03-02:
Flashing old SiK radios can be a real PITA. They are super flaky. The sync response thing is just what happens then they don't respond correctly. The only thing you can do is make sure you have a direct connection to computer via USB to radio. No hub in the middle or anything and then try again.