mavlink/qgroundcontrol #1723: Unable to reset zoom level on Analyze plot

kd0aij, 2015-07-20:
To demonstrate:
1) start QGC
2) connect
3) Select "Analyze"
4) select "xgyro" checkbox
5) zoom in then reset using mouse left and right buttons
6) try to display zacc (apparently impossible)

basil-huber, 2015-08-14:
+1: It is really annoying to close qgroundcontrol just to zoom out.
Ctrl + Mouse wheel to zoom out would be really neat

rjgritter, 2015-09-21:
+1, still an issue

kd0aij, 2016-01-22:
This issue is still present in 2.8.0-597
It means having to restart QGC if the zoom level is changed, and that can happen with a stray mouse click. I'd suggest disabling zoom completely if no better fix is possible.

DonLakeFlyer, 2016-01-22:
> I'd suggest disabling zoom completely if no better fix is possible.

I'll do that. Analyze is going to get tossed for new Qml implementation. So don't want to put any effort here.

DonLakeFlyer, 2016-01-22:
@jgoppert Do you know anything about this code? I understand nothing about the qwt stuff. Tried poking around but it's going to take me way longer to figure this out than someone who may be more familiar.

jgoppert, 2016-01-22:
I didn't write this code, the zoom bug has been there for years. @LorenzMeier might have a better idea. I know the qt c++ api pretty well and can take a look if no one else feels more comfortable with it.

DonLakeFlyer, 2016-01-23:
Ok, thanks James. I saw your name in blame, figured I'd ask.

DonLakeFlyer, 2016-02-26:
I did this:
* connect
* Select "Analyze"
* select "xgyro" checkbox
* zoomed by selecting a box with the mouse
* right-clicked with mouse and the view zoomed back out

Seems to work for me?

kd0aij, 2016-02-26:
you missed the last step: 6) try to display zacc (apparently impossible)
Wow, this issue is 7 months old now. 
I guess the analyze widget doesn't get much use.

DonLakeFlyer, 2016-02-26:
Ok, I can repro. All the additional simpler workarounds I've tried have failed. So without days of educating myself on qwt library which I know nothing about, I'm stuck until this gets tossed for something else.

DonLakeFlyer, 2016-03-13:
I thought I had tried removing zoom before. Just did it now without too much problem.

closed 2016-03-13 by DonLakeFlyer