mavlink/qgroundcontrol #1739: Critical message text unreadable outdoors (red on black)

kd0aij, 2015-07-25:
The message text popup window displays a very small font in red on a black background in both indoor and outdoor view configurations. This is impossible to read outdoors on a 1920x1080 Chromebook. The Chromebook display is very bright; similar to a Nexus 7. Dark text on a light background would be OK.

The text in the Control widget text panel is also unreadable in outdoor  ("light") mode; appears to be black on black.

DonLakeFlyer, 2015-12-03:
Different now

closed 2015-12-03 by DonLakeFlyer

kd0aij, 2015-12-05:
still dark red on black here:
[image]

info view widget timestamp is now light grey on white, very low contrast:
[image]

LorenzMeier, 2015-12-05:
@kd0aij Your state is again quite outdated. Please update.

dogmaphobic, 2015-12-05:
It may be outdated but a lot of those colors are still not optimal. I've been meaning to work on those palettes for a while but have not gotten around to it. The *Light* palette is specially hard to distinguish elements (buttons from background for example).

It will take some work as it's not just a matter of fixing the individual colors but also using the proper colors here and there. I will take a look at it soon...

kd0aij, 2015-12-05:
oops, still haven't figured out how to do a "shadow build" correctly. Thought I had rebuilt latest master, but the second try is still disappointing. @LorenzMeier are you *not* seeing the same red on black?
[image]

DonLakeFlyer, 2015-12-05:
Thought this was different now. It doesn't seem to be using the palette for the background color which is incorrect as well. Same thing for the toolbar messages at the bottom.

kd0aij, 2015-12-05:
#2356