mavlink/qgroundcontrol #1740: Add parameter tuning panel similar to APMplanner

kd0aij, 2015-07-25:
Ideally, this panel would also support configuration of the "Set to RC" function for 4 parameters at a time (one for each knob and slider on the Taranis X9). This function would be more convenient and less error prone for tuning in the field. Using "Set to RC", a single operator could safely perform inflight tuning of up to 4 parameters at a time.

These settings could be persistent and, once configured, selected from a list: e.g. FW attitude control: FW_PR_P, FW_PR_FF, FW_PR_I, FW_P_ROLLFF. The scale, init value, min and max fields for each of these parameters would all be set simply be selecting the appropriate list entry. This reduces the likelihood of errors and enables setup on the bench during log analysis and preflight preparations.

DonLakeFlyer, 2015-07-25:
Do  you have some screen shots? I don't see how this is different from the current Set RC to Param.

kd0aij, 2015-07-25:
The parameter tuning panel looks like this:
http://copter.ardupilot.com/wiki/configuration/tuning/
and has obvious advantages for quickly verifying that all the critical
parameters are correct.

The idea for "set RC" is to make it practical to preconfigure it for each
group of up to 4 parameters while doing offline log analysis, rather than
spend 10 minutes on each one at the field, with a high likelihood of
error.  One driving factor in this is that the firmware forces you to
reprogram the "set RC..." on every boot.
I originally hoped to use custom widgets for this, but you pointed out that
it can't be done that way.  Not being familiar with the QGC code, I won't
attempt to suggest an alternative at this point.

It's been suggested that the "best" way to tune is in flight using QGC,
two-way wireless telemetry and an extra person. I agree that's best, but it
obviously requires addition of long range telemetry to a fixed wing
aircraft. But then the "set RC..." function would be superfluous, and it
should instead become a knob or slider on the QGC UI for parameter tuning.

DonLakeFlyer, 2015-07-25:
Ok. I get it now. You’r going to need a corresponding request on the firmware side as well to make the rc to param settings persistent.

LorenzMeier, 2015-12-26:
@DonLakeFlyer I will provide the complete params list and text to hook up here:

LorenzMeier, 2015-12-26:
Fixed wing:

## Slider 1

Roll sensitivity

Slide to the left to make roll control faster and more accurate. Slide to the right if roll oscillates or is too twitchy.

FW_R_TC

## Slider 2

Pitch sensitivity

Slide to the left to make pitch control faster and more accurate. Slide to the right if pitch oscillates or is too twitchy.

FW_P_TC

## Slider 3

Cruise throttle

This is the throttle setting required to achieve the desired cruise speed. Most airframes have a value of 0.5-0.7.
FW_THR_CRUISE

## Slider 4

Position control sensitivity

Slide to the left to make flight in mission more accurate. Slide to the right if the flight in stabilised mode is smooth but in mission becomes twitchy.

FW_L1_PERIOD

LorenzMeier, 2015-12-26:
# Multicopter

## Slider 1

Roll sensitivity

Slide to the left to make roll control faster and more accurate. Slide to the right if roll oscillates or is too twitchy.

MC_ROLL_TC

## Slider 2

Pitch sensitivity

Slide to the left to make pitch control faster and more accurate. Slide to the right if pitch oscillates or is too twitchy.

MC_PITCH_TC

## Slider 3

Altitude control sensitivity

Slide to the left to make altitude control smoother and less twitchy. Slide to the right to make altitude control more accurate and more aggressive.
MPC_Z_FF

## Slider 4

Position control sensitivity

Slide to the left to make flight in position control mode smoother and less twitchy. Slide to the right to make position control more accurate and more aggressive.

MPC_XY_FF

LorenzMeier, 2015-12-26:
@DonLakeFlyer All these params are in PX4 master and should be good to go.

DonLakeFlyer, 2015-12-27:
Can you detail which specific MAV_TYPE values should use Copter and which specific MAV_TYPE values should use Fixed Wing? Also I assume some set of MAV_TYPE's have no tuning panel available.

DonLakeFlyer, 2015-12-27:
Also each one of these sliders needs a min/max value. Nnot necessarily min/max of parameter more than likely you bring the bounds in for the slider to only be the center part of the range. Also there needs to be a step size which are the tick marks you can move to within the range. Can you give me those as well?

DonLakeFlyer, 2015-12-27:
> Most airframes have a value of 0.5-0.7.

For cruise throttle: Tuning sliders don't show values, so this text doesn't work here.

DonLakeFlyer, 2015-12-27:
> Position control sensitivity
> 
> Slide to the left to make flight in mission more accurate. Slide to the right if the flight in stabilised mode is smooth but in mission becomes twitchy.

For fixed wing. This wording doesn't make sense to me. Label is Position Control. Maybe title should be mission something?

DonLakeFlyer, 2015-12-27:
I'll push this up, then merge in and you can tweak all you want.

DonLakeFlyer, 2015-12-27:
@LorenzMeier I merged in the tuning panel but it needs the min/max/step set for each slider. Can you finish that up?

https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/PX4/PX4TuningComponentCopter.qml#L37
https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/PX4/PX4TuningComponentPlane.qml#L37

LorenzMeier, 2015-12-30:
I finished this already, its all done.

closed 2015-12-30 by LorenzMeier

DonLakeFlyer, 2015-12-30:
Leaving this open but moving from 3.0 release. Still don't have advanced tuning page, which is what this was originally talking about.