mavlink/qgroundcontrol #1751: Ubuntu 14.04.2 - sudo required to update file location for configuation in current daily build

twoodrow, 2015-07-30:
Using sudo fixes this problem, but is this intended?

Reproduction steps:
1. Download ubuntu's daily from http://qgroundcontrol.org/downloads
2. Extract the executable
  * I used tar -xf
3. Run qgrouncontrol
  * ./qgroundcontrol
4. Observe warning message
  * note the location listed in the terminal output
5. In the file location section, add a new location
  * I used /home/qgroundcontrol-test
6. Close qgrouncontrol
7. Run qgrouncontrol
  * ./qgrouncontrol
8. Observe warning again and notice your location was not saved

DonLakeFlyer, 2015-08-23:
I assume you are talking about "File Locations" from settings. Is the new location in Settings still there when you reboot QGC?

mswingtra, 2015-08-28:
I have the same problem. @DonLakeFlyer the new location is still in Settings only if run with sudo. If I run qgroundcontrol just with ./qgroundcontrol, the location is not saved at all and 'Bad save location' error pops up. But running again with sudo afterwards works fine, previously set location is saved.

dagar, 2015-08-30:
I had this issue, then I cleared my QGC settings and can no longer reproduce.

If you need a quick fix I would suggest `rm -rf ~/.config/QGroundControl.org`, but could you check the permissions first?

    ls -ls ~/.config/QGroundControl.org

mswingtra, 2015-09-01:
When I remove QGroundControl.org folder, then on the second reboot I get following info 

    'The format for QGroundControl saved settings has been modified. Your saved settings have been reset to defaults.'

And so it is created again. It solves the problem partially though, because after I select any other folder than default one as save location, then QGC sort of works. 
However, passing meta data does not work - without flashing the custom params do not appear in default group, and after I flash custom px4 file through QGC, the default group disappears but other custom groups do not appear. The parameter.xml is generated successfully as the whole thing works perfectly, i.e. on Mac.

DonLakeFlyer, 2015-09-01:
I’ll look at the saved settings message, that seems odd.

Parameter meta data is stored in QGroundControl.org <http://qgroundcontrol.org/> folder. After you flash check to se if the version of parameter.xml from your build ends up there.

mswingtra, 2015-09-02:
So it looks as parameter meta data ends up being correctly stored in QGroundControl.org but it is not showing in QGC, see pic rel:
[image]

LorenzMeier, 2015-09-02:
Have you flashed that image *USING* QGroundControl? Because else there is no way for it to know about this XML file.

LorenzMeier, 2015-09-02:
Also I presume you have restarted QGC? We never had issues so far with XML loading, so I'm wondering if there is a particular sequence of steps breaking things.

mswingtra, 2015-09-02:
I flashed it using QGC, and I have restarted it after flashing has been completed. 
I wonder what is going on myself as it works perfectly fine for my colleagues and I am the only one with this issue.

LorenzMeier, 2015-09-02:
Did you happen to launch QGC at some point with sudo? I see very often that complete systems can get completely messed up on permissions.

mswingtra, 2015-09-02:
Yes, I am running it with sudo. 
When I had run it without sudo, I got the bad save location error. After running
    
    rm -rf ~/.config/QGroundControl.org 

rebooting, and choosing a new folder for save location, it does not complain longer about bad save location at the start. But then I cannot flash firmware as I am getting following error:

    Could not save downloaded file to /tmp/firmware.px4. Error: Permission denied
    Upgrade cancelled

LorenzMeier, 2015-09-02:
I'm closing this issue. Please never ever use sudo again for anything that is not involving changing the system configuration. If you have a permissions error you need to fix properly by giving the user the required permissions, not by running as admin. Looks like your complete permissions are now messed up on your system and you will need to fix those first.

Note that a non-writable tmp directory also means that other software you're trying to use will likely have issues, and so fixing the permissions is about fixing your Linux installation, not QGroundControl.

closed 2015-09-02 by LorenzMeier