mavlink/qgroundcontrol #1752: The IP address for a UDP connection sometimes gets lots

AndreasAntener, 2015-07-31:
When I try to connect via UDP and it doesn't work (for some reason), after I hit the disconnect button, the target IP from the connection config is lost.

dogmaphobic, 2015-07-31:
Dynamic target hosts are not persistent. Once you start QGC with an UDP link, all incoming connections become "dynamic" target hosts. When you disconnect, those target hosts are "forgotten". Once you re-enable the UDP connection and new connections come in, they are added again to the list.

The manually entered *Target Hosts* (the ones that are actually persistent) are used for redirection. Hosts you want to redirect all traffic to.

AndreasAntener, 2015-07-31:
Ok, I'm talking about the target hosts then ;), and those are the ones I loose.

dogmaphobic, 2015-07-31:
Sorry, I'm in a meeting and I can't quite look this up right now. What exactly are you doing? What hosts are you adding and why? That will help me understand it.

AndreasAntener, 2015-07-31:
Don't worry :)
- I'm adding a target host to connect to (mavros on the other end)
- Connecting
- Connection doesn't happen because network is not ready
- I hit disconnect manually
- The target host is lost

I can also have a look tomorrow maybe.

AndreasAntener, 2015-08-02:
So the problem is that the `UDPConfiguration` doesn't know if the host is dynamic or static. This is why [here](https://github.com/mavlink/qgroundcontrol/blob/68fd43e354097e13bde2eab67cd91e6fdf97284c/src/comm/UDPLink.cc#L215) my static hosts also get (temporarily) removed if `writeDatagram` fails (in my case because the host is not reachable yet). After QGC restart they are there again.

I would suggest to make the target hosts aware of their nature and only remove dynamic hosts if writing to them fails.

dogmaphobic, 2015-08-02:
> I would suggest to make the target hosts aware of their nature and only remove dynamic hosts if writing to them fails.

Yep. That's a bug. I will fix it.

Thanks for finding it! :)

dogmaphobic, 2015-08-24:
This has been fixed by #1756. We still have #1681 to deal with but this one can be closed.

closed 2015-08-24 by DonLakeFlyer