mavlink/qgroundcontrol #1753: Serial port threading broken

LorenzMeier, 2015-07-31:
When the wrong thread (I need to investigate which one) sends a MAVLink message through the serial link Qt emits this message and stops *completely* sending any data: QSocketNotifier: Socket notifiers cannot be enabled or disabled from another thread

From that point on USB doesn't receive any bytes, although the serial link API reports writing data successfully.

DonLakeFlyer, 2015-07-31:
Writes need to be on the right thread. Needs to signal itself.

LorenzMeier, 2015-07-31:
The problem is that when simulation gets activated the UI thread calls the UAS object. My first attempt at self-signalling failed, I'll have another stab.

DonLakeFlyer, 2015-07-31:
Look in PX4FirmwareUpgradeThread.cc/h or LogReplayLink.cc/h for examples of signaling to get the call on the right thread. Methods are called *OnThread.

PR #1754 (2015-08-01): UAS: Connect message sending via buffered signals
This is an attempt to fix #1753. @kd0aij If you like you can give this a go, I tested it as far as in that it enables HIL mode, but I haven't flown HIL with it yet.
closed 2015-08-01 by LorenzMeier