mavlink/qgroundcontrol #1769: QGroundControl doesn't detect Pixhawk on Arch Linux & Qt 5.5.0

Sapd, 2015-08-10:
I am trying to upgrade the firmware of the pixhawk, but it just states: "Plug in your device via USB to start firmware upgrade".

I tried:
- daily build
- stable v2.7
- compiling current git myself

System:
Arch linux 64 Bit (fully updated)
QT Version:
$ qmake
Project MESSAGE: Qt version 5.5.0
QT was installed via Pacman. All required packages were installed like the README states.

Looking into the console, QGC spams this:
FirmwareUpgradeLog: Serial Port --------------
FirmwareUpgradeLog: 	port name: "ttyACM1"
FirmwareUpgradeLog: 	description: "PX4 FMU v2.x"
FirmwareUpgradeLog: 	system location: "/dev/ttyACM1"
FirmwareUpgradeLog: 	vendor ID: 32902
FirmwareUpgradeLog: 	product ID: 39985

The vendor ID and the product ID seems to be wrong.
But if I type lsusb, it gives me this line related to the pixhawk:
Bus 001 Device 020: ID 26ac:0011 
So 26ac (hex) -> 9900 (dec) (also correct ID)
0011 (hex) -> 17 (dec) (correct ID)

So QGC seems to read out wrong Vendor/Product IDs.

HOWEVER I also tried using stable v2.5.1, this version recognizes my pixhawk and I can flash it successfully.

v01d, 2015-08-11:
Just today I had the same problem. Where did you get version 2.5.1? I see 2.5.2 and 2.5.0 on the site.

Sapd, 2015-08-11:
Oops sorry, written 2.5.1 by accident, but should be 2.5.2. 2.5.2 should work for flashing.

DonLakeFlyer, 2015-08-11:
Not sure why it’s seeing bad ids. No idea if this will help, but did you remove modem manager like it say here: https://pixhawk.org/dev/flash_px4fmu_lin?s[]=modem&s[]=manager? <https://pixhawk.org/dev/flash_px4fmu_lin?s%5B%5D=modem&s%5B%5D=manager?>

I may need to add a fall back to product name detection since the id check doesn’t seem to work in all cases.

LorenzMeier, 2015-08-11:
I think it makes sense to do so while Qt figures out their PID / VID handling. The old update code was in many ways crafted to work in all cases and it might make sense to go back to that pattern. Welcome to the cross platform war zone 8).

v01d, 2015-08-11:
In my case, I cannot remove modemmanager since it is a required dependency
of a package I use (network manager applet). The suggested udev rule is
already present in my system. Modem manager is not running either.

By the way, I can confirm that with 2.5.2 I see my board in the Firmware
tab.

DonLakeFlyer, 2015-08-11:
> The old update code was in many ways crafted to work in all cases and it might make sense to go back to that pattern.

I can't go completely back because I will lose Aerocore board detection and end up flashing Pixhawk stack onto Aerocore boards. Needs to be a final test after more specific checks fail. That should maintain all functionality.

DonLakeFlyer, 2015-08-11:
> I also modified the source code now to make it recognize my pixhawk:

Can you put together a pull request for this with the following comment:
```
            if (!found) {
                // Fall back to port name matching which could lead to incorrect board mapping. But in some cases the
                // vendor and product id do not come through correctly so this is used as a last chance detection method.
...
            }
```

Sapd, 2015-08-11:
> Not sure why it’s seeing bad ids. No idea if this will help, but did you remove modem manager like it say here:
Modemmanager was never installed on my system. Also I tried starting via sudo, but it makes no difference.
> Can you put together a pull request for this with the following comment:
Yep. https://github.com/mavlink/qgroundcontrol/pull/1770

closed 2015-08-11 by LorenzMeier