mavlink/qgroundcontrol #1772: Allow the use of the HUD when video overlay is enabled

LorenzMeier, 2015-08-12:
@dogmaphobic It would be nice if I could choose between the block instruments and retaining the normal HUD / flight instrument when video background is enabled. It doesn't make sense for a gimbal, but its great for forward-looking fixed wing.

dogmaphobic, 2015-08-14:
Yes, I thought about it too. I will see what I can do...

dogmaphobic, 2015-08-24:
I assume this can be closed? #1777 dealt with it.

DonLakeFlyer, 2015-09-08:
Assigning to me since I've changed this all around

jaxxzer, 2017-04-19:
Is this still an issue?

dogmaphobic, 2017-04-19:
I don't even remember what this was all about. With that said, being almost 2 years later, I would assume it is no longer valid as just about everything about the Fly view has changed since.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier