mavlink/qgroundcontrol #1776: RC and MavLink signal strength in Symbol like on cellphones

severinleuenberger, 2015-08-14:
Visualize the RC and Mavlink Signal strength in a small symbol (like on the cellphones) on top of QGC, near the warning-button. 

(Don Gagne asked me to open this issue). 

Here my suggestion of the Symbols: 
[image]

dogmaphobic, 2015-08-14:
The main issue is space. For RC telemetry alone, it's doable. I would probably remove the antenna symbol for simplicity's sake (the bars should be enough) but we need to identify which RSSI we're displaying.

Currently we have:

[image]

Instead of the antenna/signal symbol, we could replace with just bars:

[image]

But we would still need to identify which RSSI we're talking about.

The telemetry RSSI has two values (would need two sets of bars). For local and remote.

Should I do the changes above just for RC? Forgo the numerical value altogether and just use the bars? I can start working on a new set of icons (the one above was just a quick and dirty example).

LorenzMeier, 2015-08-14:
How about doing the numbers with a mouse over and removing them else? I don't know what 96 dB means for me anyway. Could you draw local as bars and remote as a single line / outline on top? Or should we use the min of both?

dogmaphobic, 2015-08-14:
Using the mouse hover to show values would not work on mobile (touch) devices. The idea of overlaying both local and remote telemetry is interesting but the graphics is too small for that. It becomes a smudge and you can't tell them apart.

I would think that bars alone would do fine. Do you need more granularity than 0/20/40/60/80/100?

Regarding telemetry, we need to normalize those dB values into a 0-100 scale. As they are, it's sort of useless if you don't know what is *good* or *bad*, specially given its logarithmic nature. Right now, they are only present when using the SiLabs radios (3DR). It would be good too if we captured the WiFi RSSI when using it. This is trickier because for now, the *Companion Computer* is the one handling the WiFi and relying on some UART/UDP bridge. The *radio status* would have to be fed into this bridged mavlink stream, which currently is handled by MavProxy or MavROS.

dogmaphobic, 2016-04-17:
Closing this as it has been dealt with through he drop down indicators for each item in the toolbar.

closed 2016-04-17 by dogmaphobic