mavlink/qgroundcontrol #1782: QGroundControl Doesn't Work, Again.

stevec5000, 2015-08-20:
QGC wouldn't work at all when I first tried it the middle of 2014 but eventually it was working fine in Win 7 after about 6 months of bug fixes and updates with ver.2.2.0.  However, I just tried updating to the latest ver. 2.7.1 and all I get is a black window with nothing in it and no way to do anything!
It looks like all the features are gone again, I can't see any way to program the PX4, calibrate the sensors or the R/C transmitter or connect to the aircraft!
Where can I find an older version that still works?  Thanks.

Sapd, 2015-08-21:
Looks like there is a problem on your system, because the latest version should work without any problems. Try the 2.5.2 stable http://qgroundcontrol.org/oldstable
Does this version work for you? And if not, do other QT applications work for you? (E.g. teamspeak)

LorenzMeier, 2015-08-21:
This is because Qt moved to OpenGL hardware acceleration and your graphics driver is not recent enough to support it. There will be support for old hardware in the next Qt version, which is scheduled to be released in September. Meanwhile upgrading your graphics drivers might help.

stevec5000, 2015-08-21:
So QT has screwed up everything again?   You've got to be kidding!  Win 7 has the latest updates and thankfully there is nothing else using QT.  QGC just depresses me every time I've tried to use it!

stevec5000, 2015-08-21:
Sapd, ver. 2.5.2 didn't work, ver. 2.5.0 and 2.4.0 didn't work either.  All I get is a black screen with a green button in the upper left that says Activate Engine but doesn't do anything.  Ver 2.3.0 seemed to work and I was able to update firmware and check the R/C even though I can't read most of the fuzzy orange text on the screen.  When I tried to calibrate the compass it went crazy though.

dogmaphobic, 2015-08-21:
I had not booted Windows in quite a while. I just booted it, built the current master and had no issues. I would assume if there was a real issue, a lot more people would have complained by now. There is nothing inherently wrong with Qt. I can only assume the issue is with your computer.

[image]

NaterGator, 2015-08-21:
@stevec5000 you could install the latest stable qgc and open a command prompt in the application directory then run:
```
set QT_LOGGING_RULES=qt.qpa.gl=true
qgroundcontrol.exe
```

You'll get verbose OpenGL debugging information which may help diagnose what part of your graphics stack is causing the problem.

stevec5000, 2015-08-22:
I was able to get Ver. 2.5.2 to run on a desktop with a large HD monitor so I could read what's on the screen.  It went through all the setup and calibration and seemed to all work OK but it not very convenient to carry around with a quadcopter.  Something needs to be done to make it run on a laptop again.
Then I tried Ver. 2.7 and that really has a lot of bugs in it!
I had to connect and disconnect the PX4 several times before it would find it and do the firmware update.
I tried going through the configuration but it can't find my radio!   It says: "Radio Calib Failed!  Receiver: Detected 0 radio channels, to operate PX4 you need at least 5 channels".
Note, the R/C radio works OK with Ver. 2.5.2.
Also, even though all the sensors calibrated OK it keeps playing audio saying:  "Mag self test failed, sensors need inspection".  It doesn't say what kind of inspection the senors might need!
Guess I'll go back to 2.5.2 until all the problems with 2.7 are fixed.
You guys really should try out these new versions to see if they really work before releasing anymore of them!

LorenzMeier, 2015-08-22:
They work for everybody else. What kind of USB cable are you using? Please try with a non-3DR one and on a different USB port. What you are reporting sounds like potential power issues, and the 3DR cables are unfortunately no good.

stevec5000, 2015-08-22:
I really doubt they work for "everybody else" since after ver. 2.3 text won't even display correctly on the laptop.
I don't have any 3DR cables, just generic ones, but I'll try another one just in case.

stevec5000, 2015-08-22:
Tried different cables and another desktop running Win 10.  Even tried a double ended USB cable for a hard drive to make sure there was enough power with the same results.  Still get the same sensor error and radio won't work in Ver. 2.7.1.
Guess I'll go back to Ver. 2.5.2 until all the bugs in 2.7 are fixed!

stevec5000, 2015-08-23:
Is there a drawing showing the parts layout on the PX4IO?  I found a schematic diagram but no parts layout.   I've been trying to get Qgroundcontrol to work for over a year now and finally wore out the board just turning it left and right and upside down each time to calibrate the sensors and everything.  
I doubt that it will ever work but it made a puff of smoke so now I'll have to fix it if I want to keep trying for another year!
Thanks.

closed 2015-10-25 by LorenzMeier