mavlink/qgroundcontrol #1794: Joystick input MAV_TYPE

devbharat, 2015-08-24:
The joysticks are not read properly if MAV_TYPE is not 2. See https://github.com/PX4/Firmware/issues/2755

LorenzMeier, 2015-09-01:
I need to look into this and will fix as soon as possible.

DonLakeFlyer, 2015-09-01:
I'm in the middle of re-writing the all Joystick code, so I'll fix it along with that.

DonLakeFlyer, 2015-09-01:
@LorenzMeier Seems like joystick settings should be stored by mav id not globally keyed by MAV_AUTOPILOT/MAV_TYPE?

LorenzMeier, 2015-09-01:
Well, it all seems overly complicated. There are two distinct cases:

  1) Real joystick interaction. So what is currently called "manual". This should *not* be keyed in any fashion. It should be one thing and that's it.
  2) Test modes like sending attitude or position off board control commands. This is for developers. This should be indeed keyed by MAV ID.

That said, the stuff being done in 2) should be in a different / advanced menu option and only the pure manual stuff (where button mapping is done like for a normal RC control) should be visible. All the rest is for testing purposes only and we don't want normal users to ever use it (because they're going to loose control and crash).

DonLakeFlyer, 2015-09-01:
So basically the stick calibrations, invert and button mappings are globa. Just the one mode setting value: Manual/Attitude/... is keyed by mav id. An we want to steer/hide people away from setting that mode to anything other than Manual.

LorenzMeier, 2015-09-01:
Yes, that sounds good. The testing stuff will be very popular with engineers in the Drone Linux age, but we really don't want normal users to accidentally switch it into off board mode and get confused and crash.

DonLakeFlyer, 2015-09-08:
I think this is all good now?

closed 2015-09-08 by DonLakeFlyer