mavlink/qgroundcontrol #1795: Ver. 2.7.1 Broke My PX4!

stevec5000, 2015-08-25:
When I tried ver. 2.7 it wouldn't recognize that my R/C radio was connected and the battery voltage reading at the top of the screen reads zero volts instead of 11 v. like it used to do!
I went back to the older version 2.3 and it still can't recognize the radio is connected or read battery voltage that used to work before.  
I had been using the stack configuration with a PX4IO card mounted on the FMU but to simplify trouble shooting I removed the IO card and connected everything straight to the PX4FMU.  
It still has the same problems though! 
Other functions of the FMU still work normally, the lights flash and sensors let it tilt and rotate in the flight simulator view on the map. 
I've never been able to get as far as getting all the errors cleared and trying to Arm the aircraft in any version of the program but I could at least check out all the R/C transmitter functions in older versions of the program.
Does anyone know what could be wrong with the PX4?  
Could the firmware have been corrupted or would a hardware failure cause these problems and how can I fix it?

stevec5000, 2015-08-27:
No comments?

DonLakeFlyer, 2015-10-25:
Those are issues coming from the Vehicle not QGC. Likely a firmware problem.

closed 2015-10-25 by DonLakeFlyer