mavlink/qgroundcontrol #1797: Clicking Analyse causes a crash

porcoesphino, 2015-08-26:
With version of v2.7.1 built using QT5.5 then if you click Analyse after connecting it causes a crash.

Documentation says to use QT5.4 but downloading it is tedious. To install this verision using `brew`:

```
brew install https://raw.githubusercontent.com/Homebrew/homebrew/00e46351980ad915424aad9316abffc19936809c/Library/Formula/qt5.r
```

I don't have time to add this to the readme now but please consider adding it to the OSX instructions.

Feel free to close immediately, this is just for anyone searching the issue.

DonLakeFlyer, 2015-08-26:
Qt 5.5 is not a currently supported or tested version of Qt on QGC. On OSX you cannot build QGC with WarningsAsErrors turned on (which is the default for release builds and pull requests) without it spewing out hundreds of errors. We are waiting for 5.5.1 before we move.

porcoesphino, 2015-08-27:
That's basically what I said / assumed.

I've thought about this more and there are two things that I think should be updated:

1. Add `brew` to the OSX Install section of the README.md

    - It seems to be the prevailing package manager and spelling out how to install 5.4 speeds up the time it takes a new developer to join the project.

2. Make `qmake` hard fail if the wrong version of QT is present (maybe with a flag to override)

    - Currently, when you run `qmake` it says that the project is using private interfaces and it is tied to a specific version of QT. Confusingly, there is a QT version on the line above that message but it is not the version the project needs, it is the version that `qmake` was installed from. The output of `qmake` has no mention of the required QT version.

If I get some free time I'll put together a pull request but I'm surprisingly busy now. Sorry.

DonLakeFlyer, 2015-08-27:
The readme already tell you how to install Qt on OSX. There are likely many different ways to do it. Going directly to the Qt website is the most direct way that is available as soon as the releases become available. I'd rather keep things simple, otherwise I have to spend too much time digging around to update a complex readme.

For #2: Something I've been wanting to do for a while but have never gotten around to it.

LorenzMeier, 2015-09-01:
@DonLakeFlyer I have heard that doing this without a vehicle connected happens on the latest daily as well. Can you repro this?

DonLakeFlyer, 2015-09-01:
This was likely the Info View widget and I fixed it a few days back.

LorenzMeier, 2015-09-07:
@porcoesphino Closing due to likely being fixed. Please reopen if it persists.

closed 2015-09-07 by LorenzMeier