mavlink/qgroundcontrol #1819: Cut over plotting to QML Charts

LorenzMeier, 2015-09-01:
@dogmaphobic @DonLakeFlyer I stumbled across this, which seemed intriguing:
http://doc.qt.io/QtCharts/qtcharts-qmloscilloscope-example.html

There is also this, but the official Qt solution might be preferable due to probably better rendering backends.

dogmaphobic, 2015-09-01:
Hehe... I had found that a long time ago and I was salivating at the time. But if you scroll all the way down, you will notice that's only available for a *paid* license :)

DonLakeFlyer, 2015-09-01:
Yup I saw that as well. I was looking at this: http://qcustomplot.com

LorenzMeier, 2015-09-01:
I don't think it differs from the normal Qt license. I'm pretty confident its available the same way.

DonLakeFlyer, 2015-09-01:
It's only in the paid license. You can get a 30 day free trial, then you have to pay. There's a chart somewhere that says what is in what. I can't seem to find it now.

dogmaphobic, 2015-09-01:
> I can't seem to find it now.

Me either. I've seen it before too. Their normal paid license is $350/month. I wonder what kind of arrangement could be made for an open source project though...

LorenzMeier, 2015-09-01:
If you follow the link at the bottom of the page it tells you that you can download for free.

DonLakeFlyer, 2015-09-01:
This is what I see at the bottom:
[image]
Click that, and then you can see how you can get a 30 day free trial on the commercial license. Only Open Source is free, and it's not in that version.

NaterGator, 2016-02-16:
The plotting interface is still a big sore point. What if we consider something like Chart.js integration:
https://github.com/jwintz/qchart.js

[image]

dogmaphobic, 2016-02-16:
> What if we consider something like Chart.js integration

I've looked at that and considered it. Qt Chart is now supposedly open source. Waiting to see how that goes before investing time with JavaScript.

NaterGator, 2016-02-16:
Ah, missed that... the release is [here](http://blog.qt.io/blog/2016/01/13/new-agreement-with-the-kde-free-qt-foundation/)

It looks like it will be GPLv3, is that going preclude us from using it on mobile platforms?

LorenzMeier, 2016-02-16:
No. It just means we need a proper license for the App Store build.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier