mavlink/qgroundcontrol #1825: Qt5.4 ppa is missing packages

projectskye, 2015-09-02:
The README for the prerequisits for Ubuntu 14.04 is broken. The packages qt54creator and qt54qbs do not exist (anymore?): [[https://launchpad.net/~beineri/+archive/ubuntu/opt-qt541-trusty]]

A workaround is to simply run "apt-get install qtcreator" and manually adjust the compiler to the 5.4 version of qmake. However, this is not a really nice solution.
I'm not sure about what qt54qbs is for.

kjeremy, 2015-09-28:
:+1: Seeing this as well

DonLakeFlyer, 2015-09-28:
@dagar Do you know how to fix this?

dogmaphobic, 2015-09-28:
If you are going to be running Qt Creator, you are probably better off installing Qt using the Qt Online Installer (http://www.qt.io/download-open-source). It gives you by far more control of what to install and it's easier to maintain. Not to mention that it's self contained. It can coexist with anything else you may have installed.

dagar, 2015-09-28:
No, the PPA is out of our control. For now we can remove qt54creator and qt54qbs from the dependencies. It'll be nice once we can use the default Qt provided by the system.

DonLakeFlyer, 2015-09-28:
> It'll be nice once we can use the default Qt provided by the system.

We'll likely always come in and out of phase with that. We need some set of instructions or process that will work when we're out of phase. I don't know enough about Linux stuff to know what this is. Hopefully it's possible somehow.

DonLakeFlyer, 2015-10-09:
readme updated

closed 2015-10-09 by DonLakeFlyer