mavlink/qgroundcontrol #1826: New map engine: Tile loading issues when fully zoomed out

LorenzMeier, 2015-09-02:
It seems like when I'm fully zoomed out (seeing the whole globe) it cannot load the full map. There are other artefacts when zooming in, and the console shows these:

```
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
NULL Map request reply
tile request error  "Problem with tile image"
tile request error  "Problem with tile image"
NULL Map request reply
NULL Map request reply
```

DonLakeFlyer, 2015-10-09:
@dogmaphobic Didn't you get rid of these?

dogmaphobic, 2015-10-12:
This was a Qt bug, which I reported and it has been fixed for 5.5.x. I bumped into it when I had enabled the map to follow the vehicle (as opposed to always north). For that to work, I needed a map large enough to rotate within the viewing window. That's why I disabled that option waiting for the 5.5.x fix. The bug would still be present (even with always north) when used on very large screens though.

Here: https://bugreports.qt.io/browse/QTBUG-45508

closed 2015-10-25 by DonLakeFlyer