mavlink/qgroundcontrol #1830: travis-ci and appveyor issues

dagar, 2015-09-04:
Catch all for remaining work and any issues with travis-ci and appveyor.

- [ ] linux unittests
- [ ] match teamcity build options
- [ ] windows builds (appveyor)
- [ ] windows installer
- [ ] mac package
- [ ] daily or continuous builds (deploy to s3?)
- [ ] release builds (PR #1805 then deploy to s3 or start using github releases?)
- [ ] gstreamer?
- [ ] ubuntu dpkg + ppa

dagar, 2015-09-04:
Teamcity qmake

qmake qgroundcontrol.pro -r -spec linux-g++-64 CONFIG+=debug CONFIG+=WarningsAsErrorsOn
qmake qgroundcontrol.pro -r -spec linux-g++-64 CONFIG+=release CONFIG+=WarningsAsErrorsOn
qmake qgroundcontrol.pro -r -spec macx-clang CONFIG+=debug CONFIG+=WarningsAsErrorsOn
qmake qgroundcontrol.pro -r -spec macx-clang CONFIG+=release CONFIG+=WarningsAsErrorsOn
qmake.exe -r -tp vc CONFIG-=debug_and_release CONFIG+=debug CONFIG+=WarningsAsErrorsOn qgroundcontrol.pro
qmake.exe -r -tp vc CONFIG-=debug_and_release CONFIG+=release CONFIG+=WarningsAsErrorsOn qgroundcontrol.pro

Installers
qmake qgroundcontrol.pro -r -spec linux-g++-64 CONFIG+=installer
qmake qgroundcontrol.pro -r -spec macx-clang CONFIG+=installer
qmake -r -tp vc CONFIG+=installer qgroundcontrol.pro

dogmaphobic, 2015-09-04:
> gstreamer?

Yes, once the builds are producing video streaming enabled binaries, the runtime libraries will have to be installed as well. For now, video streaming is not being built by the *official* build. You need to build it on your own. There are no special flags required to add video support. If the build systems finds the appropriate development libraries installed, it will produce a version of QGC with video streaming support. That however, requires the runtime libraries to be installed. If you build it on your own, you will already have it. For distribution however, we need to handle it.

The libraries for the various platforms are defined here: [VideoStreaming ](https://github.com/mavlink/qgroundcontrol/tree/master/src/VideoStreaming)

I only know for sure on Mac OS (GStreamer Framework, which is a separate installer) and Ubuntu (GStreamer runtime installed using apt-get, which can be easily handled by having it as a dependency). I have to see what the requirements are for the other platforms (I don't remember)

closed 2015-10-15 by dagar