mavlink/qgroundcontrol #1833: Controllers menu disapeared from Settings

jlbilou93, 2015-09-04:
hello I have just built version 41035462b9e764abb0d79293271139921db2b59a
and there is no controllers menu in the settings.
On 4415f32dff32558c14a897abb8fe67dd197a7d43 it seems ok.

DonLakeFlyer, 2015-09-04:
Controllers menu? Can you be more specific? There never was a "Controllers" menu.

DonLakeFlyer, 2015-09-06:
There is no such thing.

closed 2015-09-06 by DonLakeFlyer

jlbilou93, 2015-09-22:
In menu File -> Settings -> The tab Controllers between Comm Links and MAVLink disappeared.
How do we configure Joystick now?
On the binary version v2.7.1 I can see this tab.
On master of today 22nd of September don't see it anymore. Is it a special build option?
I am building on Ubuntu 15.04 with QT v5.5.0.

LorenzMeier, 2015-09-22:
Just use the setup perspective like for the rest of the system. There is a new pane for the joystick there.

jlbilou93, 2015-09-22:
Thx for your answer. Maybe my drone is not in the correct mode?
What are the drone requirements to access the Setup configuration?
- MAV_STATE_STANDBY MAV_MODE_*_DISARMED ? 
- MAV_MODE_FLAG_MANUAL_INPUT_ENABLED ?

What are the minimal mavlink messages I should send to qgroundcontrol ?
Currently I am sending:
HEARTBEAT
ATTITUDE
SYSTEM_STATUS
RAW_GPS
GLOBAL_POSITION_INT
MISSION_CURRENT

DonLakeFlyer, 2015-09-22:
Joystick support if specific to the PX4 Flight Stack firmware implementation. It doesn’t require a specific mode to be in. It required a joystick to be connected before it will show up.

jlbilou93, 2015-09-23:
That is too bad, on previous versions e.g 2.7.1, I could make my drone use joystick inputs with GENERIC autopilot.

DonLakeFlyer, 2015-09-24:
Hmm. That should still be doable but it will take a little work.

DonLakeFlyer, 2015-09-25:
What are you using for firmware? It doesn’t need any additional parameters set when you use a joystick?

jlbilou93, 2015-09-26:
I am using a Parrot Bebop firmware.
On version 2.7.1 I didn't need additional parameters, because we could set the axis and the value were between -1000 and 1000.