mavlink/qgroundcontrol #1839: Vehicle heading invalid until GPS lock

dogmaphobic, 2015-09-05:
Adding this here so it won't fall through the cracks.

The code showing the vehicle position (and *heading*) is causing an error when there is no GPS lock. The heading is invalid and causes an error in the QML code. There should be some check and only add a vehicle to the map if its position is valid (which only happens when you have a GPS lock).

This code: https://github.com/mavlink/qgroundcontrol/blob/master/src/FlightMap/FlightMap.qml#L178

DonLakeFlyer, 2015-09-06:
Yup. Had this floating around in the back of my head but hadn't gotten to it yet.

LorenzMeier, 2015-09-06:
Note there are two headings: Course over ground, which is from GPS and should only be set if the velocity is greater than ~3 m/s. And magnetic heading, which comes from the attitude estimate and is always correct. I would recommend going with the yaw value if ground speed is < 3 m/s and with the course over ground if its larger.

DonLakeFlyer, 2015-09-08:
Right now the heading is pulled from the UAS attitudeChanged signal. This signal is output from multiple mavlink messages:
- MAVLINK_MSG_ID_ATTITUDE
- MAVLINK_MSG_ID_ATTITUDE_QUATERNION
- MAVLINK_MSG_ID_VFR_HUD (only if it doesn't get it from one of the above sources)

Are the first two message exclusive from each other. Wondering if the signal ends stomping on each other from these two messages?

DonLakeFlyer, 2015-09-09:
The GPS lock problem is fixed. If you the heading value is being pulled from the wrong place can you add a new issue?

closed 2015-09-09 by DonLakeFlyer