mavlink/qgroundcontrol #1840: Ver. 2.7.1 Unable to Read Battery Voltage!

stevec5000, 2015-09-06:
Ver 2.7.1 is not able to read the battery voltage on the PX4 or on a Pixhawk, saying 0.0 volts.
Even though the Pixhawk is running off the battery and has the 6 pin cable connected to the Power connector it still can't read the voltage on Pin 4.  The voltage on Pin 4 of the Power connector is 1.1 volts but the battery voltage always says 0.0 volts.
It reads 0.0 volts on the PX4 with the battery connected to the Power input connector on the PX4IO even though the buzzer is going off and it uses that input for power.

stevec5000, 2015-09-07:
Isn't anyone fixing problems or maintaining this software anymore?

DonLakeFlyer, 2015-10-25:
If battery is not coming through then something is wrong on the vehicle side

closed 2015-10-25 by DonLakeFlyer