mavlink/qgroundcontrol #1845: Joystick reading hangs with UI while changing views on QGCS

devbharat, 2015-09-06:


DonLakeFlyer, 2015-09-25:
We're going to need to change the threading model to fix this. Each Vehicle needs to be on it's own thread amoung other things.

devbharat, 2015-10-08:
Is the fix fro this a priority soon ? I think it's pretty major issue while doing sitl, we could also temporally push the joystick data through python etc to free up use of qgcs for more important tasks...

DonLakeFlyer, 2015-10-08:
Not a the top of the list at this point. Typically threading changes to QGC have taken a long time to to settle out and be bug free. So I'm waiting until the latest round of work is complete and we can cut a Stable.

devbharat, 2015-10-08:
Thanks for the update. Maybe a small script to externally forward the manual control packets could be patch for the while.

DonLakeFlyer, 2015-10-08:
I'm curious as to how this is such a high priority. Maybe I'm not understanding the problem. How can a 1/10th of a second or so of delay in sending a joystick message, be any different than losing a few packets over radio. If the Vehicle can't handle that, how can it fly safely?

devbharat, 2015-10-08:
It lasts longer...sort of like slightly over a second. Especially when switching to the analysis tab for the first time after starting the simulation. Probably this is so much more than you'd expect as the system is already under almost full load.

DonLakeFlyer, 2015-10-08:
If you switch to all tabs before you fly. You should be able to shorten these delays by quite a bit as a workaround. Are you running on windows?

devbharat, 2015-10-08:
Ah! Should try that. No, ubuntu.

LorenzMeier, 2015-10-12:
@DonLakeFlyer The tabs also load slowly on mobile (2-6 seconds) it might make sense to force them being pre-loaded since right now it feels pretty laggy.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier