mavlink/qgroundcontrol #1850: Jostick calibration not responding

tumbili, 2015-09-08:
When I begin the calibration I'm told to move the throttle stick in the upper position. After doing so the calibration routine does not respond, the throttle stick in the GUI never changes position. The sliders show me that the channels from the Joystick are read properly.

DonLakeFlyer, 2015-09-08:
Is this the XBox stick Lorenz was telling me about? If so, I showed him how to do some debugging to figure out what is going on.

Specifically turn on JoystickConfigControllerLog.debug=true to see what is going on.

LorenzMeier, 2015-09-08:
@DonLakeFlyer Is the ini debugging documented somewhere? I searched the wiki:
http://qgroundcontrol.org/start?do=search&id=ini

and the README with no success:
https://github.com/mavlink/qgroundcontrol

The README just links to the Qt docs and mentioned a non-existent file at the top level of the repo. It would be good to have this made much more explicit, because right now even I don't know how to get the logs you want to you.

DonLakeFlyer, 2015-09-08:
I'll update the wiki: http://doc.qt.io/qt-5/qloggingcategory.html

DonLakeFlyer, 2015-09-08:
If you delete your qtlogging.ini file. It will get recreated with all the logging categories pre-populated so you can just set true/false on what you want.

DonLakeFlyer, 2015-09-08:
http://www.qgroundcontrol.org/bugreporting

LorenzMeier, 2015-09-24:
@DonLakeFlyer Here is a log file:

```
0
"MAVLINK_INSPECTOR_DOCKWIDGET"
Orphaned log file count 0
Temp log "/tmp/FlightData019893.mavlink"
Unknown element in XML:  "board"
src/uas/UASWaypointManager.cc 1092 "Requesting waypoint list..."
TYPE CHANGED TO: 2
Unknown message from system: 1 message: 87
Unknown message from system: 1 message: 245
2
Unknown message from system: 1 message: 76
Incorrect mode sequence request:active 2 1
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 0 0 0 0 false
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 1 -32768 32768 0 false
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 2 -32768 32768 0 false
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 3 -32768 32768 0 false
JoystickConfigControllerLog: _setHelpImage "calibration/joystick/joystickCenter.png"
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 0 0 0 0 false
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 1 -32768 32768 0 false
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 2 -32768 32768 0 false
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 3 -32768 32768 0 false
JoystickConfigControllerLog: _setHelpImage "calibration/joystick/joystickCenter.png"
JoystickConfigControllerLog: _calSaveCurrentValues
JoystickConfigControllerLog: _saveAllTrims trim 0
JoystickConfigControllerLog: _saveAllTrims trim 0
JoystickConfigControllerLog: _saveAllTrims trim 0
JoystickConfigControllerLog: _saveAllTrims trim 0
JoystickConfigControllerLog: _saveAllTrims trim 0
JoystickConfigControllerLog: _setHelpImage "calibration/joystick/joystickThrottleUp.png"
JoystickConfigControllerLog: _calSaveCurrentValues
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 4 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
JoystickConfigControllerLog: _stickSettleComplete starting settle timer, _stickDetectValue:value 0 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 4 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 4 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 4 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 4 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 4 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 4 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
[... 1118 lines cut ...]
JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 2 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 3 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 4 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 0 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 1 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 2 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 3 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 4 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 0 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 1 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 2 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 3 0
JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 4 0
[... cut ...]
JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 4 0
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 0 0 0 0 false
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 1 -32768 32768 0 false
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 2 -32768 32768 0 false
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 3 -32768 32768 0 false
JoystickConfigControllerLog: _setHelpImage "calibration/joystick/joystickCenter.png"
After app delete
```

DonLakeFlyer, 2015-09-25:
Can you try the pull. Looks like your joystick has more than 4 axes and currently joystick config only supports up to 4 axes. Can you run with JoystickManagerLog=true for me and post the output. It will report the axis count.

LorenzMeier, 2015-09-26:
@tumbili Is this now resolved and working?

LorenzMeier, 2015-09-30:
@devbharat Could you try master? It should work for you without hacks.

devbharat, 2015-09-30:
Let me try it and reply here tomorrow.

devbharat, 2015-10-01:
I briefly tried the latest master on qgc, with the frsky joystick. It recognized the channels correctly and the calibration routine also worked fine. But when I tried to arm,  I couldn't. I observed

- the rc channel0 value wasn't visible on the plot on the ananlysis tab in qgc
- when I printed the manual_setpoint.z in the code to read the throttle stick value, it actually changed when I yawed, not when throttling, even you the calibration gui on qgcs seemed to recognize sticks correctly.

I didn't get to check it very thorough, but I have a feeling something is not quite right there.

DonLakeFlyer, 2015-10-01:
What mode were you using? Default mode does not emulated RC channels.

devbharat, 2015-10-05:
I tried with the latest qgc and latest firmware again. I am using NORMAL mode(no other mode seems to push to manual_sp). I calibrated the joystick, shows correctly when I shift sticks on calibration GUI. The Issue is:
- Yaw seems to be on throttle and throttle on yaw (I did calibrate twice, correctly!)
- As explained above, throttle up down changes manual.r, which only takes -1 or +1 value
- x and y are correct, except that at the very end of the stick motion, manual_sp changes sign! (If it was -1 at almost full stick, at full stick it becomes +1!)

DonLakeFlyer, 2015-10-05:
Are you viewing manual_sp in the analyze view to look at values?

LorenzMeier, 2015-10-05:
Roman debugged this a bit and it happens after SDL but before its being sent to the vehicle. The root cause is calibration going wrong. For him yaw has min / max values of zero.

LorenzMeier, 2015-10-05:
The range wrap is a result of the trigonometric functions getting inputs bigger than 1 / -1, which is also a result of the failed calibration.

devbharat, 2015-10-06:
@DonLakeFlyer No, I checked it out of qgc, in the autopilot firmware.

DonLakeFlyer, 2015-10-06:
Can you send a log from calibration? —logging:JoystickConfigControllerLog

devbharat, 2015-10-08:
I started QGC with the additional —logging:JoystickConfigControllerLog

Is the FlightData.mavlink the logging file you are looking for ?

DonLakeFlyer, 2015-10-08:
No. Logging comes out on the console. On Windows this will be in a separate window that shows up when you run. On OS X/Ubuntu it will be in the Terminal output.

devbharat, 2015-10-09:
Here it is:

    nknown message from system: 1 message: 76
    libpng warning: iCCP: known incorrect sRGB profile
    JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 0 0 0 0 false
    JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 1 -32768 32767 0 false
    JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 2 -32768 32767 0 false
    JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 3 -32768 32767 0 false
    JoystickConfigControllerLog: _setHelpImage "calibration/joystick/joystickCenter.png"
    JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 0 0 0 0 false
    JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 1 -32768 32767 0 false
    JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 2 -32768 32767 0 false
    JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 3 -32768 32767 0 false
    JoystickConfigControllerLog: _setHelpImage "calibration/joystick/joystickCenter.png"
    JoystickConfigControllerLog: _calSaveCurrentValues
    JoystickConfigControllerLog: _saveAllTrims trim 0
    JoystickConfigControllerLog: _saveAllTrims trim 0
    JoystickConfigControllerLog: _saveAllTrims trim 0
    JoystickConfigControllerLog: _saveAllTrims trim 0
    JoystickConfigControllerLog: _setHelpImage "calibration/joystick/joystickThrottleUp.png"
    JoystickConfigControllerLog: _calSaveCurrentValues
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 1 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 2 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 3 0
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 3 0 0
[... 1478 lines cut ...]
    JoystickConfigControllerLog: _inputStickDetect function:axis:value 1 2 32767
    JoystickConfigControllerLog: _inputStickDetect settle complete, function:axis:value 1 2 32767
[... cut ...]
    JoystickConfigControllerLog: _setHelpImage "calibration/joystick/joystickCenter.png"
    JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 0 0 0 0 false
    JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 1 -32768 32767 0 false
    JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 2 -32768 32767 0 false
    JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 3 -32768 32767 0 false
    ^[[1;2A^[[1;2A^[[1;2A^[[1;2A^[[1;2A^[[1;2A^[[1;2A^[[1;2A^[[1;2A

DonLakeFlyer, 2015-10-09:
I haven't quite figured it out yet. The logging has some holes in it which need to be fixed for me to get full data. The one thing I noticed is that when you calibrate Yaw, the Throttle stick is still full down. Throttle needs to be centered when calibrating Yaw. The code doesn't enforce this at the moment since it assume spring centered joystick on both sticks. I'll fix that. That may becausing the problem but I'm not sure yet.

You can see at the very end of the log that Throttle (axis 0) has bad values:
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 0 0 0 0 false
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 1 -32768 32767 0 false
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 2 -32768 32767 0 false
JoystickConfigControllerLog: Read settings axis:min:max:trim:reversed 3 -32768 32767 0 false

DonLakeFlyer, 2015-10-09:
@devbharat Can you try this again using the very latest daily build. It has better logging to help diagnose this. Also use --logging:JoystickLog,JoystickConfigControllerLog on command line.

devbharat, 2015-10-09:
Can you tell the commit hash ?

devbharat, 2015-10-09:
21:00:(master) ./qgroundcontrol --logging:JoystickLog,JoystickConfigControllerLog
    Filter rules "JoystickLog.debug=true
    JoystickConfigControllerLog.debug=true
    "
    Saved files location "/home/bharat/Documents/QGroundControl"
    JoystickLog: _loadSettings  "FrSky FrSky Taranis Joystick"
    JoystickLog: _loadSettings calibrated:throttlemode:badsettings true 1 false
    JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 0 0 0 0 false false
    JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 1 -32768 32767 0 false false
    JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 2 -32768 32767 0 false false
    JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 3 -32768 32767 0 false false
    JoystickLog: _loadSettings function:axis:badsettings 0 1 false
    JoystickLog: _loadSettings function:axis:badsettings 1 2 false
    JoystickLog: _loadSettings function:axis:badsettings 2 3 false
    JoystickLog: _loadSettings function:axis:badsettings 3 0 false
    JoystickLog: _loadSettings button:action 0 ""
    JoystickLog: _loadSettings button:action 1 ""
    JoystickLog: _loadSettings button:action 2 ""
    JoystickLog: _loadSettings button:action 3 ""
    JoystickLog: _loadSettings button:action 4 ""
    JoystickLog: _loadSettings button:action 5 ""
    JoystickLog: _loadSettings button:action 6 ""
    JoystickLog: _loadSettings button:action 7 ""
    JoystickLog: _loadSettings button:action 8 ""
    JoystickLog: _loadSettings button:action 9 ""
    JoystickLog: _loadSettings button:action 10 ""
    JoystickLog: _loadSettings button:action 11 ""
    "9px;" "13px;" "21px;"
    0
    "PRIMARY_FLIGHT_DISPLAY_DOCKWIDGET"
    Orphaned log file count 1
    Orphaned log file "/tmp/FlightData019893.mavlink"
    ^C
    21:02:(master) ./qgroundcontrol --logging:JoystickLog,JoystickConfigControllerLog
    Filter rules "JoystickLog.debug=true
    JoystickConfigControllerLog.debug=true
    "
    Saved files location "/home/bharat/Documents/QGroundControl"
    JoystickLog: _loadSettings  "FrSky FrSky Taranis Joystick"
    JoystickLog: _loadSettings calibrated:throttlemode:badsettings true 1 false
    JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 0 0 0 0 false false
    JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 1 -32768 32767 0 false false
    JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 2 -32768 32767 0 false false
    JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 3 -32768 32767 0 false false
    JoystickLog: _loadSettings function:axis:badsettings 0 1 false
    JoystickLog: _loadSettings function:axis:badsettings 1 2 false
    JoystickLog: _loadSettings function:axis:badsettings 2 3 false
    JoystickLog: _loadSettings function:axis:badsettings 3 0 false
    JoystickLog: _loadSettings button:action 0 ""
    JoystickLog: _loadSettings button:action 1 ""
    JoystickLog: _loadSettings button:action 2 ""
    JoystickLog: _loadSettings button:action 3 ""
    JoystickLog: _loadSettings button:action 4 ""
    JoystickLog: _loadSettings button:action 5 ""
    JoystickLog: _loadSettings button:action 6 ""
    JoystickLog: _loadSettings button:action 7 ""
    JoystickLog: _loadSettings button:action 8 ""
    JoystickLog: _loadSettings button:action 9 ""
    JoystickLog: _loadSettings button:action 10 ""
    JoystickLog: _loadSettings button:action 11 ""
    "9px;" "13px;" "21px;"
    0
    "PRIMARY_FLIGHT_DISPLAY_DOCKWIDGET"
    Orphaned log file count 1
    Orphaned log file "/tmp/FlightData019893.mavlink"
    Temp log "/tmp/FlightData019893.mavlink"
    Unknown element in XML:  "board"
    src/uas/UASWaypointManager.cc 1092 "Requesting waypoint list..."
    TYPE CHANGED TO: 2
    Unknown message from system: 1 message: 87
[... 2451 lines cut ...]
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 2 32767
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 3 0
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 0 -338
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 1 0
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 2 30065
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 3 0
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 0 -338
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 1 0
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 2 27025
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 3 0
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 0 -338
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 1 0
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 2 21282
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 3 0
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 0 -338
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 1 0
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 2 15539
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 3 0
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 0 -338
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 1 0
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 2 13512
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 3 0
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 0 -338
    JoystickConfigControllerLog: _inputStickMin function:axis:value 1 1 0
    JoystickConfigControllerLog: _inputStickMin function

DonLakeFlyer, 2015-10-09:
Ok, i see it now. The initial set is good:
> JoystickLog: _saveSettings name:axis:min:max:trim:reversed "FrSky FrSky Taranis Joystick" 0 -32768 32767 -32768 false
> 

But then it gets overwritten by bad values:
> JoystickLog: _saveSettings name:axis:min:max:trim:reversed "FrSky FrSky Taranis Joystick" 0 0 0 -10811 false
> 
Not sure why. I think I may be able to replicate your setup. How do you use you FrSky as a joystick. Do you just plug in via USB and it shows up automatically. Or do you need to do something else to set it up.

> On Oct 9, 2015, at 12:09 PM, Bharat Tak <notifications@github.com> wrote:
> 
> 21:00:(master) ./qgroundcontrol --logging:JoystickLog,JoystickConfigControllerLog
> Filter rules "JoystickLog.debug=true
> JoystickConfigControllerLog.debug=true
> "
> Saved files location "/home/bharat/Documents/QGroundControl"
> JoystickLog: _loadSettings "FrSky FrSky Taranis Joystick"
> JoystickLog: _loadSettings calibrated:throttlemode:badsettings true 1 false
> JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 0 0 0 0 false false
> JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 1 -32768 32767 0 false false
> JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 2 -32768 32767 0 false false
> JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 3 -32768 32767 0 false false
> JoystickLog: _loadSettings function:axis:badsettings 0 1 false
> JoystickLog: _loadSettings function:axis:badsettings 1 2 false
> JoystickLog: _loadSettings function:axis:badsettings 2 3 false
> JoystickLog: _loadSettings function:axis:badsettings 3 0 false
> JoystickLog: _loadSettings button:action 0 ""
> JoystickLog: _loadSettings button:action 1 ""
> JoystickLog: _loadSettings button:action 2 ""
> JoystickLog: _loadSettings button:action 3 ""
> JoystickLog: _loadSettings button:action 4 ""
> JoystickLog: _loadSettings button:action 5 ""
> JoystickLog: _loadSettings button:action 6 ""
> JoystickLog: _loadSettings button:action 7 ""
> JoystickLog: _loadSettings button:action 8 ""
> JoystickLog: _loadSettings button:action 9 ""
> JoystickLog: _loadSettings button:action 10 ""
> JoystickLog: _loadSettings button:action 11 ""
> "9px;" "13px;" "21px;"
> 0
> "PRIMARY_FLIGHT_DISPLAY_DOCKWIDGET"
> Orphaned log file count 1
> Orphaned log file "/tmp/FlightData019893.mavlink"
> ^C
> 21:02:(master) ./qgroundcontrol --logging:JoystickLog,JoystickConfigControllerLog
> Filter rules "JoystickLog.debug=true
> JoystickConfigControllerLog.debug=true
> "
> Saved files location "/home/bharat/Documents/QGroundControl"
> JoystickLog: _loadSettings "FrSky FrSky Taranis Joystick"
> JoystickLog: _loadSettings calibrated:throttlemode:badsettings true 1 false
> JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 0 0 0 0 false false
> JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 1 -32768 32767 0 false false
> JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 2 -32768 32767 0 false false
> JoystickLog: _loadSettings axis:min:max:trim:reversed:badsettings 3 -32768 32767 0 false false
> JoystickLog: _loadSettings function:axis:badsettings 0 1 false
> JoystickLog: _loadSettings function:axis:badsettings 1 2 false
> JoystickLog: _loadSettings function:axis:badsettings 2 3 false
> JoystickLog: _loadSettings function:axis:badsettings 3 0 false
> JoystickLog: _loadSettings button:action 0 ""
> JoystickLog: _loadSettings button:action 1 ""
> JoystickLog: _loadSettings button:action 2 ""
> JoystickLog: _loadSettings button:action 3 ""
> JoystickLog: _loadSettings button:action 4 ""
> JoystickLog: _loadSettings button:action 5 ""
> JoystickLog: _loadSettings button:action 6 ""
> JoystickLog: _loadSettings button:action 7 ""
> JoystickLog: _loadSettings button:action 8 ""
> JoystickLog: _loadSettings button:action 9 ""
> JoystickLog: _loadSettings button:action 10 ""
[... 796 lines cut ...]
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 3 5742
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 0 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 1 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 2 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 3 6756
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 0 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 1 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 2 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 3 8107
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 0 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 1 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 2 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 3 8783
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 0 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 1 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 2 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 3 10134
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 0 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 1 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 2 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 3 12161
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 0 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 1 0
> JoystickConfigControllerLog: _inputStickDetect function:axis:value 2 2 0
> JoystickConfigControllerLog: _inputStickDetect func

devbharat, 2015-10-09:
The firmware shipped by frsky is a fork of opentx, usb joystick function didn't really work with that (as far as I remember). So I flashed the original openTX, it works with that. Just power on the radio and then stick the USB in, it will show up as joystick. 

OpenTX companion is used to flash the firmware.

LorenzMeier, 2015-10-09:
Ours (later models from TBS) did work out of the box. Thanks for the hint!

DonLakeFlyer, 2015-10-11:
Ok, now I'm really confused as to why this is trashing the values for axis 0. Taranis is working fine for me. Just tried again to make sure. I'm going to put in a special pull which has a ton of logging in it to figure out where th values are changing.

DonLakeFlyer, 2015-10-11:
For debugging:
This is the start of the save process: https://github.com/mavlink/qgroundcontrol/blob/master/src/VehicleSetup/JoystickConfigController.cc#L518

I'm pretty sure the values are good before this. Then somewhere during save they are getting trashed. You can see in this first log entry that axis 0 is correct:
JoystickLog: _saveSettings name:axis:min:max:trim:reversed "FrSky FrSky Taranis Joystick" 0 -32768 32767 -32768 false

But then after that it gets screwed up:
JoystickLog: _saveSettings name:axis:min:max:trim:reversed "FrSky FrSky Taranis Joystick" 0 0 0 -10811 false

LorenzMeier, 2015-10-11:
@devbharat @tumbili One of you will need to put this one to rest.

tumbili, 2015-10-12:
Unfortunately, I cannot even test this because I'm experiencing another problem now.
I can calibrate the Joystick in QGC but I do not get any response in the "Radio" tab. The problem seems to be that there are no rc messages comming from the PX4 side. I have put a printf in the mavlink receiver, the message appears at times but then is not visible anymore.
This happens with latest master as well as with another branch which I have successfully used in the past to fly with Joystick. I have rebuilt QGC serveral times, including fresh clones.

LorenzMeier, 2015-10-12:
@DonLakeFlyer Roman found out that the joystick only works in the joystick setup view. I found some suspicious QML code in there:

```
    JoystickConfigController {
        id:             controller
        factPanel:      panel
        statusText:     statusText
        cancelButton:   cancelButton
        nextButton:     nextButton
        skipButton:     skipButton

        Component.onCompleted: {
            controllerCompleted = true
            if (rootQGCView.completedSignalled) {
                controllerAndViewReady = true
                controller.start()
            }
        }
    }

    onCompleted: {
        if (controllerCompleted) {
            controllerAndViewReady = true
            controller.start()
            updateAxisCount()
        }
    }
```

Is this really just the calibration controller or is this actually related to the interface to the vehicle?

LorenzMeier, 2015-10-12:
@DonLakeFlyer This here could be failing. The joystick thread is not running on Linux:
https://github.com/mavlink/qgroundcontrol/blob/d242549ac14c5d40058ea91d12374a9e037707b8/src/Vehicle/Vehicle.h#L139

The property is supposed to be stored in the settings?

tumbili, 2015-10-12:
@DonLakeFlyer I tried hardcoding the calibration values for the yaw axis where they are saved here 
https://github.com/mavlink/qgroundcontrol/blob/master/src/Joystick/Joystick.cc#L157

I verified that I now only get this log:
JoystickLog: _saveSettings name:axis:min:max:trim:reversed "FrSky FrSky Taranis Joystick" 0 -32768 32767 -32768 false

But still in the end the calibration for yaw is corrupted.

tumbili, 2015-10-12:
@DonLakeFlyer Also I tried to hardcode the values where they are normally loaded here https://github.com/mavlink/qgroundcontrol/blob/master/src/Joystick/Joystick.cc#L94

No success.

LorenzMeier, 2015-10-12:
So we have a hotfix in, but its unclear where the real issue is originating. Almost looks like some sort of buffer overflow spilling into index 0 on Linux: #1960.

LorenzMeier, 2015-10-20:
@DonLakeFlyer We've found it: The bug is that _axisMax is hardcoded to 4. There are two max axis counts:

  * One is the maximum number of mapped axes. This can be hardcoded to 4
  * One is the maximum number of axes of the current joystick. This has to be dynamic.

Because those are not properly separated any joystick with an axis greater than the 4th will fail. E.g. this code here makes that assumption:
https://github.com/mavlink/qgroundcontrol/blob/dca59af3b8da77027ffe70a6a82d7c2f8c059d36/src/VehicleSetup/JoystickConfigController.cc#L148-L150

The whole Joystick controller needs to be refactored to use two max values, and it should check that the max number of dynamic axes is < than the allocated memory size to hold axes.

@tumbili Your segfault likely is the result of that structure being too small and code missing bounds checking.

DonLakeFlyer, 2015-10-20:
> and it should check that the max number of dynamic axes is < than the allocated memory size to hold axes.

That is pretty much what it does right now. It just has the size locked to 4. If there is a bug with static axis count, the same bug will be there with dynamic axis count. It will just be less likely to hit since it would be less likely for them to be out of sync.

DonLakeFlyer, 2015-10-20:
Enter too early! There has been two bugs seen, one is a segfault. Catch that in the debugger and it should be trivial to see what is going on. That could likely be an array bound issue. The other is calibration values getting trashed after calibration is complete. This one is quite strange and isn't really explained by an array bounds problem. Especially since the bug logs I've seen for that happen on throttle which was in array element 0.

DonLakeFlyer, 2016-01-21:
Still a problem?

MaEtUgR, 2019-07-24:
I guess that's completely outdated and can be closed. Right @RomanBapst ?

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier