mavlink/qgroundcontrol #1858: Even if full param load fails we should still show Parameter Editor

DonLakeFlyer, 2015-09-09:


robert-budde, 2015-10-06:
Hi!

I would like to second this!

Two threads on the google newsgroups which might be related to this:
- https://groups.google.com/d/msg/qgroundcontrol/DdFtszlkCig/Mr1AMyAzS9gJ
- https://groups.google.com/d/msg/qgroundcontrol/m8snYXop4oQ/kDJKQEHg8EgJ

If you point me in the the right direction I might be able to assist/fix the problem.

Best regards,
Robert

DonLakeFlyer, 2015-10-06:
It's fairly involved to fix since I want to provide a new signal from the plugin to indicate a failed parameter load. I'll move this up in priority. That said, this error does mean something is wrong with your firmware. You should also try to figure out what is going wrong there.

DonLakeFlyer, 2015-10-06:
Ask and ye shall receive.

DonLakeFlyer, 2015-10-07:
Fixed

closed 2015-10-07 by DonLakeFlyer