mavlink/qgroundcontrol #1859: Create virtual on-screen joystick

LorenzMeier, 2015-09-09:
@DonLakeFlyer I ordered a tablet and I'm eager to get things going on mobile. It seems this is very high on our wish list:
http://kunalmaemo.blogspot.ch/2011/11/virtual-joystick-with-qml.html

I wouldn't want to disrupt your mission planning spree with it, but maybe after that? The wifi module in the Pixhawk 2 kit should allow you to fly a small quad just using Wifi then. @dogmaphobic does this feel interesting?

dogmaphobic, 2015-09-09:
With a single *stick*? I'm not sure you could implement two simultaneous *sticks* though. We would have to come up with something clever to deal with it.

LorenzMeier, 2015-09-09:
It looks like we're covered:
http://doc.qt.io/qt-5/qml-qtquick-multipointtoucharea.html

LorenzMeier, 2015-09-09:
This looks like its a complete implementation:
https://github.com/mana/manamobile/blob/master/example/qml/main/Joystick.qml

It uses a touch point which has to be inside a multipointtoucharea. It also looks like they already added mouse support to ease testing.

dogmaphobic, 2016-05-06:
Don has done this quite a while ago.

closed 2016-05-06 by dogmaphobic