mavlink/qgroundcontrol #1879: Camera setup entry required in config menu

LorenzMeier, 2015-09-23:
We need to allow the user to configure the bottom and forward looking cameras. In particular for the Snapdragon flight.

DonLakeFlyer, 2015-09-25:
I filled out the form to hopefully get one of these sent to me. Is there some info somewhere on what needs to be done?

LorenzMeier, 2015-09-25:
I can implement it if you provide me with an empty menu entry for the cameras which has the QML for an image display widget. I just need to then copy the image from the UAS object to the widget.

DonLakeFlyer, 2015-09-25:
Is this the mavlink based video thing which is needed for PX4 Flow as well? That disappeared somehow. I can bring it back.

LorenzMeier, 2015-09-25:
Yes, thats the one we need.

DonLakeFlyer, 2015-09-25:
Do you remember what that was called? I went back to 2013 and I can't seem to find it anywhere in mainwindow.

DonLakeFlyer, 2015-09-25:
And of course I found it as soon as I typed that. I went back to far!

DonLakeFlyer, 2015-09-25:
What should we call this thing so that it doesn't get confused with video streaming support?

LorenzMeier, 2015-09-25:
I would call it camera configuration and stick it into the normal setup menus, *not* into the main UI. We then would add multiple QML image views there for the different cameras.

DonLakeFlyer, 2015-09-25:
How will multiple images be requested to come across mavlink. I don't anything in data_transmission_handshake to specify one or more sources.

LorenzMeier, 2015-09-25:
It will be replaced. We can start with one image view and we'll need a button which sets the video only mode on the flow sensor.

DonLakeFlyer, 2015-09-25:
Can we set up a time tomorrow to talk through this? A little busy now, but want to clear some things up.

LorenzMeier, 2015-09-26:
If you are around within the next hour yes, after that its difficult.

LorenzMeier, 2015-11-06:
@DonLakeFlyer More people in need of this now.

LorenzMeier, 2015-11-06:
@DonLakeFlyer New images are announced using this:
https://github.com/mavlink/qgroundcontrol/blob/master/src/uas/UAS.cc#L992

And you can fetch the image using this:
https://github.com/mavlink/qgroundcontrol/blob/master/src/uas/UAS.cc#L1352

this is a LOW RATE LINK! So no OpenGL needed. Just draw the QImage on a canvas and everybody will be happy. This should not take more than an hour.

dogmaphobic, 2015-11-07:
Same as in #403. It's *experimentally* there. Let me know how it goes...

DonLakeFlyer, 2015-12-03:
Do you need something other than the PX4 Flow config?

DonLakeFlyer, 2016-01-10:
@LorenzMeier Is this done now?

LorenzMeier, 2016-01-10:
Yes, all done.

closed 2016-01-10 by LorenzMeier