mavlink/qgroundcontrol #1880: QGC crashed when click "Analyze" - Ubuntu 14.04

rentt, 2015-09-24:
After connect to the pixhawk and click the "Analyze", the app crashed. Both tested with debug and release version. The QGC is built with QT Creator with the latest code from master branch.
BTW, the other features like "setup", "Plan" and "Fly" works well.

OS(installed on VMWare): Linux ubuntu 3.16.0-49-generic #65~14.04.1-Ubuntu SMP Wed Sep 9 10:03:23 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

Compiler: QT Creator 3.4.2, Based on Qt 5.5.0 (GCC 4.9.1 20140922 (Red Hat 4.9.1-10), 64 bit)

Any debug files should I post?

DonLakeFlyer, 2015-09-25:
Can you repro this with a posted daily build? Not sure if we're running into a problem with using Qt 5.5 which isn't quite officially supported by QGC yet.

rentt, 2015-10-05:
I have tried the daily build. First the app could not lunch because it require the QT libraries(my QT installed at /home/Qt5.5.0). 
"./qgroundcontrol: error while loading shared libraries: libQt5Location.so.5: cannot open shared object file: No such file or directory"

After set the env variable, the qgroundcontrol lunched successfully. "Analyze" button worked fine, while "Fly" button not worked(crashed after click).
"tom@ubuntu:~/Downloads/QGroundControl-Pixhawk$ ./qgroundcontrol
QDir( "/home/tom/.config/QtProject" , nameFilters = { "*" },  QDir::SortFlags( Name | IgnoreCase ) , QDir::Filters( Dirs|Files|Drives|AllEntries ) )
Saved files location "/home/tom/Documents/QGroundControl"
JoystickManagerLog: Available joysticks
JoystickManagerLog: 	none found
"9px;" "13px;" "21px;"
0
""
Orphaned log file count 0
"PARAMETER_INTERFACE_DOCKWIDGET,FILE_VIEW_DOCKWIDGET"
./qgroundcontrol: symbol lookup error: ./qgroundcontrol: undefined symbol: _ZN16QGeoTiledMapDataC1EP29QGeoTiledMappingManagerEngineP7QObject
"

BTW, I tried install the Qt54 libraries required in the README.md, it seems some libraries source not available.

DonLakeFlyer, 2015-10-06:
Are you running QGC with Qt 5.5 or Qt 5.4? QGC isn’t verified on Qt 5.5 yet, so you will likely run into problems.

rentt, 2015-10-12:
Thanks for the information. I am running QGC with QT5.5.

closed 2015-10-12 by rentt

ChristophTobler, 2015-11-20:
@rentt I have the same error: "./qgroundcontrol: error while loading shared libraries: libQt5Location.so.5: cannot open shared object file: No such file or directory".  What did you do to fix that? What is set the env variable? thx

DonLakeFlyer, 2015-11-20:
https://github.com/mavlink/qgroundcontrol/pull/2249 <https://github.com/mavlink/qgroundcontrol/pull/2249>