mavlink/qgroundcontrol #1890: GStreamer pipeline settings via UI

mhkabir, 2015-09-26:
The Gstreamer pipeline elements need to be exposed via the UI in order to support all kinds of video feeds. 
The relevant elements are here:
https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/VideoReceiver.cc#L82

They can be set to whatever Gstreamer supports, but that is rather sytem-specific since it depends on installed plugins. A simple text field would work for now I'm sure. It should reset to default after throwing an error incase the particular element isn't supported.

dogmaphobic, 2015-09-26:
The current settings are the optimal set for a “**it just works**” scenario. The libraries and plugins have been pruned to just the minimum necessary. Anything different would require adding libraries and/or plugins to the distribution. This is specially complicated for Android. Just about any other video format will increase latency, which makes FPV impractical.

With that said, I’m not averse to allowing a *developer* to change these settings for their own needs. I just strongly oppose making this a *user* setting. This is way beyond what a normal user would know what to do. The beauty of Open Source is that it allows you full access to the source so you can do whatever change you want, if you know what you are doing. Presenting the average user every option under the sun will only make something that should just work into something only a CS PhD can use.

mhkabir, 2015-09-26:
Hmm, yes. But with the current settings, you're locked to the
assumption that another GST instance is streaming the video data (and
you have control over it.) Also, you're assuming that on the air side
the encoding is being done by GST, etc.

In my case, I have a dedicated compression engine running which
encodes from HDMI to a RTSP data stream(h.264 compression) Been using
Gstreamer on the ground to just view it (rtspsrc) Having this small
flexibility to chose the source would be nice :)

dogmaphobic, 2015-09-26:
> I have a dedicated compression engine running which
> encodes from HDMI to a RTSP data stream

What is this system? I've been looking for something like it for quite a while. Something that would allow video from the GoPro (or whatever is hanging off the gimbal) to be streamed as well. In that case, you would have a choice of flight camera (static) and video camera (gimbal).

As to configurability, what do you have in mind?

mhkabir, 2015-09-27:
Eh. Sorry I cannot tell you about it exactly due to client confidentiality,
but you can Google "TI DM368" for information. ;)

Configuration -  I personally just need a RTSP source. The rest should work
like normal since the video is H.264 encoded.
Unrelated question -  Do you know if the QT image sink is hardware
accelerated on Android ? And what sort of latency do you get on Android,
and at what resolution/framerate?

LorenzMeier, 2015-09-27:
@mhkabir We're not supporting proprietary solutions on our own time. Your vendor can contact us and propose how he thinks he can support the community. Otherwise we will focus on the development partners who do support the community, e.g. Qualcomm.

closed 2015-09-27 by LorenzMeier

mhkabir, 2015-09-28:
@LorenzMeier I'm not asking you to support a proprietary solution. All I'm asking is for a way to expose the pipeline parameters.
Personally, I'm using IP cameras e.g (http://www.ti.com/tool/DM368IPNC-MT5) for video, which stream a RTSP video stream. So is @vooon. This is a matter of simply supporting one more type of camera out of the box, and you/the community can use it fully as well without much effort. I had planned to make a wiki page out of the thing, once working. 
Indeed, this is the camera I'm using : http://www.alibaba.com/product-detail/wdr-ip-module-TI-Davinci-DM368_60238936009.html?spm=a2700.7724857.35.1.aTW3w8
See the front of the copter.
[image]

Kiwa21, 2016-05-11:
@mhkabir Did you succeed in opening a RTSP stream into QGC ? I also have a commercial IP camera and I am trying to connect it to QGC but I am very confused... I open to any suggestions

jaxxzer, 2019-06-27:
Update, QGC does work with RTSP streams. It does not depend on that rtsp stream to be generated by gstreamer. It works with off the shelf ip cameras, for example, and it should work with your chipset @mhkabir.