mavlink/qgroundcontrol #1891: Disappearing map tiles on Android

mhkabir, 2015-09-26:
Map tiles load fine, but some tiles in the map background then disappear. Zooming in/out or moving the map around gets them back, but some other one disappears then.

[image]

DonLakeFlyer, 2015-09-26:
Can you upload the screen shot again. You need to wait until it's finished uploading before saving I think.

mhkabir, 2015-09-26:
Done.

dogmaphobic, 2015-09-26:
There is a lot of logging being set to the console during tile retrieval. It would help to see the logs to see what's going on.

mhkabir, 2015-09-27:
@dogmaphobic Where can I find them?

dogmaphobic, 2015-09-27:
On Android, it should be out to logcat. With that said, don't worry about it for the time being. I'm rewriting the caching mechanism, which deals with some of that.

dogmaphobic, 2015-09-27:
Hmm, there seems to be an issue with Google Satellite maps. Some tiles are getting 404 responses. I've been wondering how far we could get on this free ride (google, in theory, does not like people doing what we're doing). Switch to some other map type and see if the error persists. I'm done with caching but this is a separate issue.

DonLakeFlyer, 2015-09-27:
I've noticed that. It's only been happening recently. Used to work just fine. I don' think it's something we changed on our side. Likely a change on the google side. Does the old map pull satellite tiles without problem?

dogmaphobic, 2015-09-27:
It's definitely a change on the Google side. I'm researching as we "speak". The issue seems to be isolated to google satellite maps. Everything else seems to still be working fine.

mhkabir, 2015-09-27:
I don't quite think my issue is the same as yours. All the tiles on mine load fine, and then disappear after a few seconds. I'll try the other map engines meanwhile to see if it's any different

dogmaphobic, 2015-09-27:
If they show up and then disappear, it's definitely something else. Once I get this done, I will do an Android build and see if I can find anything...

dogmaphobic, 2015-09-27:
I just did an Android build and it's all working as it should. Not sure what you are seeing though. If the tiles disappear, it sounds more like an OpenGL/Texture paint issue. What device are you using?

mhkabir, 2015-09-27:
Its a Nexus 7 2012 running 5.1.1 
This has been happening since I first started using QGC on Android, so it isn't something new. I'm inclined to believe that its a paint issue, because the thing looks like a glitch in a game ;)

mhkabir, 2015-09-28:
Happens with all map providers.
[image]

mhkabir, 2015-11-14:
@dogmaphobic @DonLakeFlyer
Guys, this still happens, and is a lot worse nowadays (like 1/4th of the map is visible). I'm running Android 6 and the Play Store builds, and this is so bad that it is practically unflyable.

It'd be great if you could have a look into the rendering issue.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier