mavlink/qgroundcontrol #1907: Parameter search doesn't work with multiple component ids

DonLakeFlyer, 2015-09-29:


CestMoiMartin, 2016-09-01:
hi !

Maybe a relationship : the MAVLink Parameter Protocol seems to be not respected with multiple components :
 - When QGC send a PARAM_REQUEST_LIST message, I send back all my parameters for each component with a PARAM_VALUE message. 
 - Then QGC send a PARAM_REQUEST_READ for each parameter with a wrong component id.

Martin

DonLakeFlyer, 2016-09-02:
There is a bug in on that already. Although I hadn't figure out what was going wrong yet. Looks like you've detailed that here.

DonLakeFlyer, 2016-09-04:
@CestMoiMartin Can you provide a QGC console ParameterLoaderLog for the problem you describe above? Instructions here: https://donlakeflyer.gitbooks.io/qgroundcontrol-user-guide/content/Support/Support.html

CestMoiMartin, 2016-09-05:
> content/Support/Support.html

Hi !

In the description.txt file, you can find a textual description of the
parameters structure with components id and the global index.
I generated 2 tests with 2 and 3 components with different QGC behaviors.

I hope those files will help you to find this bug.
The UAV is a fully home-made drone because I wanted to learn about drone
conception. Maybe I made mistakes in Mavlink implementations, but I can
find bug in my part. Then don't blame me if finally the misunderstanding is
on my side... :-)

King regards
Martin

TEST WITH 3 COMPONENTS
======================
compid  =  30
  Kp_Acc   idx 0
  Ki_Acc   idx 1
  Kp_Comp   idx 2
  Kp_Press   idx 3
  Ki_Press   idx 4
compid  =  10
  kp_angleP   idx 5
  kp_angleQ   idx 6
  kp_angleR   idx 7
compid  =  11
  pPID_Kp   idx 8
  pPID_Ki   idx 9
  pPID_Imax   idx 10
  qPID_Kp   idx 11
  qPID_Ki   idx 12
  qPID_Imax   idx 13
  rPID_Kp   idx 14
  rPID_Ki   idx 15
  rPID_Imax   idx 16

With this structure, QGC request values for compid=10  from idx 0 to 13.

TEST WITH 2 COMPONENTS
======================
compid  =  30
  Kp_Acc   idx 0
  Ki_Acc   idx 1
  Kp_Comp   idx 2
  Kp_Press   idx 3
  Ki_Press   idx 4
compid  =  10
  kp_angleP   idx 5
  kp_angleQ   idx 6
  kp_angleR   idx 7

 With this structure, QGC request values 
    for compid=10  from idx 0 to 4.
	for compid=30  from idx 5 to 7

CestMoiMartin, 2016-09-05:
Hummmm.... I forgot the console ouput generated....

If I understood well, the QGC should not request values one-by-one because
I already sent all values.

Regards
Martin

[D] at :0 - "Settings location "C:/Users/Martin/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[D] at :0 - "Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n""
[D] at :0 - "Map Cache in: "C:/Users/Martin/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db""
[!] at :0 - "VideoReceiver::start() failed. Error with gst_element_factory_make('udpsrc')"
[!] at :0 - "VideoReceiver::start() failed"
[D] at :0 - "Orphaned log file count 0"
[E] at qrc:qml/OfflineMap.qml:350 - "qrc:qml/OfflineMap.qml:350: TypeError: Cannot read property 'complete' of null"
[D] at :0 - "Temp log "C:/Users/Martin/AppData/Local/Temp/FlightData444976.mavlink" false"
[E] at :0 - "All parameters missing, unable to determine default componet id"
[E] at qrc:///QtQuick/Controls/Private/MenuContentItem.qml:178 - "qrc:///QtQuick/Controls/Private/MenuContentItem.qml:178: ReferenceError: Acccessible is not defined"
[E] at qrc:///QtQuick/Controls/Private/MenuContentItem.qml:178 - "qrc:///QtQuick/Controls/Private/MenuContentItem.qml:178: ReferenceError: Acccessible is not defined"
[D] at :0 - "Temp log "C:/Users/Martin/AppData/Local/Temp/FlightData444976.mavlink676762" false"
[E] at :0 - "The following parameter indices could not be loaded after the maximum number of retries:  "8, 9, 10, 11, 12, 13, 14, 15, 16""
[D] at :0 - "Temp log "C:/Users/Martin/AppData/Local/Temp/FlightData444976.mavlink245672" false"
[E] at :0 - "The following parameter indices could not be loaded after the maximum number of retries:  "10, 11, 12, 13, 14, 15, 16""
[D] at :0 - "Temp log "C:/Users/Martin/AppData/Local/Temp/FlightData444976.mavlink937084" false"
[E] at :0 - "The following parameter indices could not be loaded after the maximum number of retries:  "0, 1, 2, 3, 4, 8, 9, 10, 11, 12, 13, 14, 15, 16, 0, 1, 2, 3, 4, 5, 6, 7, 10, 11, 12, 13, 14, 15, 16, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15, 16""
[D] at :0 - "Temp log "C:/Users/Martin/AppData/Local/Temp/FlightData444976.mavlink804318" false"
[D] at :0 - "QGCMessageBox (unit testing) "File Error" "Could not create file.\nPlease provide a different file name to save to.""
[D] at :0 - "Temp log "C:/Users/Martin/AppData/Local/Temp/FlightData444976.mavlink322625" false"
[D] at :0 - "Temp log "C:/Users/Martin/AppData/Local/Temp/FlightData444976.mavlink627589" false"
[E] at :0 - "The following parameter indices could not be loaded after the maximum number of retries:  "0, 1, 2, 3, 4, 5, 6, 7""
[E] at qrc:qml/OfflineMap.qml:350 - "qrc:qml/OfflineMap.qml:350: TypeError: Cannot read property 'complete' of null"

CestMoiMartin, 2016-09-05:
Grrr.... I didn't activate logs...
This is not my day !!!

[D] at :0 - "Settings location "C:/Users/Martin/AppData/Roaming/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[D] at :0 - "Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\nMAVLinkProtocolLog.debug=true\nParameterLoaderLog.debug=true\nParameterLoaderVerboseLog.debug=true\n""
[D] at :0 - "Map Cache in: "C:/Users/Martin/AppData/Local/cache/QGCMapCache300" / "qgcMapCache.db""
[!] at :0 - "VideoReceiver::start() failed. Error with gst_element_factory_make('udpsrc')"
[!] at :0 - "VideoReceiver::start() failed"
[D] at :0 - "Orphaned log file count 0"
[D] at :0 - "Temp log "C:/Users/Martin/AppData/Local/Temp/FlightData334665.mavlink" false"
[D] at :0 - "Default component param """
[D] at :0 - "Request to refresh all parameters for component ID: "MAV_COMP_ID_ALL""
[D] at :0 - "_parameterUpdate (id: 80 componentId: 30 name: "Kp_Acc" count: 8 index: 0 mavType: 9 value: QVariant(float, 1) )"
[D] at :0 - "Seeing component for first time, id: 30 parameter count: 8"
[D] at :0 - "_waitingReadParamIndexMap: QMap((1, 0)(2, 0)(3, 0)(4, 0)(5, 0)(6, 0)(7, 0))"
[D] at :0 - "_waitingReadParamNameMap QMap()"
[D] at :0 - "_waitingWriteParamNameMap QMap()"
[D] at :0 - "waitingReadParamIndexCount: 7"
[D] at :0 - "totalWaitingParamCount: 7"
[D] at :0 - "Adding new fact"
[D] at :0 - "_parameterUpdate (id: 80 componentId: 30 name: "Ki_Acc" count: 8 index: 1 mavType: 9 value: QVariant(float, 0.01) )"
[D] at :0 - "_waitingReadParamIndexMap: QMap((2, 0)(3, 0)(4, 0)(5, 0)(6, 0)(7, 0))"
[D] at :0 - "_waitingReadParamNameMap QMap()"
[D] at :0 - "_waitingWriteParamNameMap QMap()"
[D] at :0 - "waitingReadParamIndexCount: 6"
[D] at :0 - "totalWaitingParamCount: 6"
[D] at :0 - "Adding new fact"
[D] at :0 - "_parameterUpdate (id: 80 componentId: 30 name: "Kp_Comp" count: 8 index: 2 mavType: 9 value: QVariant(float, 0) )"
[D] at :0 - "_waitingReadParamIndexMap: QMap((3, 0)(4, 0)(5, 0)(6, 0)(7, 0))"
[D] at :0 - "_waitingReadParamNameMap QMap()"
[D] at :0 - "_waitingWriteParamNameMap QMap()"
[D] at :0 - "waitingReadParamIndexCount: 5"
[D] at :0 - "totalWaitingParamCount: 5"
[D] at :0 - "Adding new fact"
[D] at :0 - "_parameterUpdate (id: 80 componentId: 30 name: "Kp_Press" count: 8 index: 3 mavType: 9 value: QVariant(float, 0.5) )"
[D] at :0 - "_waitingReadParamIndexMap: QMap((4, 0)(5, 0)(6, 0)(7, 0))"
[D] at :0 - "_waitingReadParamNameMap QMap()"
[D] at :0 - "_waitingWriteParamNameMap QMap()"
[D] at :0 - "waitingReadParamIndexCount: 4"
[D] at :0 - "totalWaitingParamCount: 4"
[D] at :0 - "Adding new fact"
[D] at :0 - "_parameterUpdate (id: 80 componentId: 30 name: "Ki_Press" count: 8 index: 4 mavType: 9 value: QVariant(float, 0.01) )"
[D] at :0 - "_waitingReadParamIndexMap: QMap((5, 0)(6, 0)(7, 0))"
[D] at :0 - "_waitingReadParamNameMap QMap()"
[D] at :0 - "_waitingWriteParamNameMap QMap()"
[D] at :0 - "waitingReadParamIndexCount: 3"
[D] at :0 - "totalWaitingParamCount: 3"
[D] at :0 - "Adding new fact"
[D] at :0 - "_parameterUpdate (id: 80 componentId: 10 name: "kp_angleP" count: 8 index: 5 mavType: 9 value: QVariant(float, 2.5) )"
[D] at :0 - "Seeing component for first time, id: 10 parameter count: 8"
[D] at :0 - "_waitingReadParamIndexMap: QMap((0, 0)(1, 0)(2, 0)(3, 0)(4, 0)(6, 0)(7, 0))"
[D] at :0 - "_waitingReadParamNameMap QMap()"
[D] at :0 - "_waitingWriteParamNameMap QMap()"
[D] at :0 - "waitingReadParamIndexCount: 10"
[D] at :0 - "totalWaitingParamCount: 10"
[D] at :0 - "Adding new fact"
[D] at :0 - "_parameterUpdate (id: 80 componentId: 10 name: "kp_angleQ" count: 8 index: 6 mavType: 9 value: QVariant(float, 2.5) )"
[D] at :0 - "_waitingReadParamIndexMap: QMap((0, 0)(1, 0)(2, 0)(3, 0)(4, 0)(7, 0))"
[D] at :0 - "_waitingReadParamNameMap QMap()"
[D] at :0 - "_waitingWriteParamNameMap QMap()"
[D] at :0 - "waitingReadParamIndexCount: 9"
[D] at :0 - "totalWaitingParamCount: 9"
[D] at :0 - "Adding new fact"
[D] at :0 - "_parameterUpdate (id: 80 componentId: 10 name: "kp_angleR" count: 8 index: 7 mavType: 9 value: QVariant(float, 2.5) )"
[D] at :0 - "_waitingReadParamIndexMap: QMap((0, 0)(1, 0)(2, 0)(3, 0)(4, 0))"
[D] at :0 - "_waitingReadParamNameMap QMap()"
[D] at :0 - "_waitingWriteParamNameMap QMap()"
[D] at :0 - "waitingReadParamIndexCount: 8"
[D] at :0 - "totalWaitingParamCount: 8"
[D] at :0 - "Adding new fact"
[... 72 lines cut ...]
[D] at :0 - "Read re-request for (componentId: 10 paramIndex: 0 retryCount: 10 )"
[D] at :0 - "Read re-request for (componentId: 10 paramIndex: 1 retryCount: 10 )"
[D] at :0 - "Read re-request for (componentId: 10 paramIndex: 2 retryCount: 10 )"
[D] at :0 - "Read re-request for (componentId: 10 paramIndex: 3 retryCount: 10 )"
[D] at :0 - "Read re-request for (componentId: 10 paramIndex: 4 retryCount: 10 )"
[D] at :0 - "Read re-request for (componentId: 30 paramIndex: 5 retryCount: 10 )"
[D] at :0 - "Read re-request for (componentId: 30 paramIndex: 6 retryCount: 10 )"
[D] at :0 - "Read re-request for (componentId: 30 paramIndex: 7 retryCount: 10 )"
[D] at :0 - "Giving up on (componentId: 10 paramIndex: 0 retryCount: 11 )"
[D] at :0 - "Giving up on (componentId: 10 paramIndex: 1 retryCount: 11 )"
[D] at :0 - "Giving up on (componentId: 10 paramIndex: 2 retryCount: 11 )"
[D] at :0 - "Giving up on (componentId: 10 paramIndex: 3 retryCount: 11 )"
[D] at :0 - "Giving up on (componentId: 10 paramIndex: 4 retryCount: 11 )"
[D] at :0 - "Giving up on (componentId: 30 paramIndex: 5 retryCount: 11 )"
[... cut ...]
[D] at :0 - "Gave up on initial load after max retries (componentId: 10 paramIndex: 3 )"
[D] at :0 - "Gave up on initial load after max retries (componentId: 10 paramIndex: 4 )"
[D] at :0 - "Gave up on initial load after max retries (componentId: 30 paramIndex: 5 )"
[D] at :0 - "Gave up on initial load after max retries (componentId: 30 paramIndex: 6 )"
[D] at :0 - "Gave up on initial load after max retries (componentId: 30 paramIndex: 7 )"
[E] at :0 - "The following parameter indices could not be loaded after the maximum number of retries:  "0, 1, 2, 3, 4, 5, 6, 7""

DonLakeFlyer, 2016-09-13:
@CestMoiMartin It seems like the vehicle may not be responding to the parameter protocol correctly. The PARAM_VALUE.param_count value should be the count of params for the specific component, not the total count of all params across all components. Same thing for PARAM_VALUE.param_index, this is an index for the component specific list, not the full list.

DonLakeFlyer, 2016-09-19:
@CestMoiMartin Any update on this?

CestMoiMartin, 2016-09-20:
Hi Don !

sorry, I didnt work on my project.
I have a problem: I can launch the QGroundControl.... then I can't test the
protocol with the right indexes.... :-(

I keep in touch with you.
Thanks for your answer
Martin

CestMoiMartin, 2016-09-20:
Thanks to a new install, I can run QGC...

After the initial transmission of all parameters, the QGC requests some
missing (?) parameters.
Then I can see on QGC the parameters according my component structure.

I can say that it seems to works now !!
I have to proceed some additional tests (for param set protocol).

Unfortunatly, my trouble didnt resolve your initial trouble.

Thanks for your help.
Martin

DonLakeFlyer, 2016-09-20:
From you posts I can't tell if you are still having a problem or not. It seems like you are running you own version of firmware and that firmware has not implemented the parameter protocol correctly. Are you running a standard firmware?

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier