mavlink/qgroundcontrol #1910: Debug Assertion Failed!

zshn25, 2015-10-01:
Hi, I tried building the debug built on my Windows 10 64-bit laptop. The built successfully completed after I tried to fix many errors. Now, when I run it, it shows the following error. Please help!!
[image]
When I pressed Ignore button, this window opens
[image]
The code for this error is this
```
{
    Element const* const last = string + maximum_count;
    Element const*       it   = string;
    for (; it != last && *it != '\0'; ++it)        //having the Break-point here and it is highlighting '\0'
    {
    }
    return static_cast<size_t>(it - string);
}
```

DonLakeFlyer, 2015-10-01:
You must be doing something wrong along the way if you needed to fix any errors. The windows version builds cleanly with no warnings. First step is to follow the instructions exactly as specified here: https://github.com/mavlink/qgroundcontrol <https://github.com/mavlink/qgroundcontrol>, VS 2015 is not yet verified to use with QGC and may likely be the issue.

zshn25, 2015-10-02:
Thanks Don, using 2013 VS solved all the problems.

DonLakeFlyer, 2015-10-07:
Looks like we have work to do to move to 2015!

closed 2015-10-07 by DonLakeFlyer

zshn25, 2016-05-04:
Hey mavlink/qgroundcontr,

Discover high quality jobs and connect with recruiters on LinkedIn.

- Zeeshan

Accept invite: https://www.linkedin.com/e/v2?e=0-ins5rfgt-ed&t=ssuw&ek=invite_guest&li=29&m=invite_guest&ts=india_a&sharedKey=qmmAJl5Q&invitationId=6133430003435663360

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=0-ins5rfgt-ed&t=lun&midToken=AQFmrFVhGEd36Q&ek=invite_guest&loid=AQGLLHJtVtZh4wAAAVR5Sm-WHGfsmX9akAb16RgEkRJQ2PAA4O9xSQ&eid=0-ins5rfgt-ed

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=0-ins5rfgt-ed&a=customerServiceUrl&ek=invite_guest

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

zshn25, 2016-05-10:
Zeeshan Khan would like to connect on LinkedIn. How would you like to respond?

Accept: https://www.linkedin.com/e/v2?e=-jrzliy-io16bina-v4&a=preRegInvite&ek=first_guest_reminder_01&li=14&m=hero&ts=accept_text&sharedKey=qmmAJl5Q&invitationID=6133430003435663360

View Zeeshan Khan&#39;s profile: https://www.linkedin.com/e/v2?e=-jrzliy-io16bina-v4&a=preRegInvite&ek=first_guest_reminder_01&li=3&m=hero&ts=profile_text&sharedKey=qmmAJl5Q&invitationID=6133430003435663360

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=-jrzliy-io16bina-v4&t=lun&midToken=AQFmrFVhGEd36Q&ek=first_guest_reminder_01&li=16&m=unsub&ts=HTML&eid=-jrzliy-io16bina-v4&loid=AQG9qbB1CyJf_gAAAVSaxJUyxMpT-Zj_2qUKJtcSZ48xnT2cz-35rxt6mYRFbPlhqIrY1cQzICl6WHoHT8VYeFxrO1_PyZtLv_V7PWlVxM5F8Imkx-5yPNPMWQPrv9YbgAjul1ARxEXVmIFYz1xUs7yRj0gznDzKMAmqJhfO2CIK1ftUk9UhO-SNsakWDv-4Nquh

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=-jrzliy-io16bina-v4&a=customerServiceUrl&ek=first_guest_reminder_01

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.

dogmaphobic, 2016-05-10:
> Zeeshan Khan would like to connect on LinkedIn. How would you like to respond?

@zshn25 Could you please stop posting Likedin requests here?

zshn25, 2016-05-17:
Zeeshan Khan would like to connect on LinkedIn. How would you like to respond?

Accept: https://www.linkedin.com/e/v2?e=-jrzliy-iob6c77x-m9&t=ssuw&ek=second_guest_reminder_01&li=13&m=hero&ts=accept_text&sharedKey=qmmAJl5Q&invitationId=6133430003435663360

View Zeeshan Khan&#39;s profile: https://www.linkedin.com/e/v2?e=-jrzliy-iob6c77x-m9&t=ssuw&ek=second_guest_reminder_01&li=3&m=hero&ts=profile_text&sharedKey=qmmAJl5Q&invitationId=6133430003435663360

You received an invitation to connect. LinkedIn will use your email address to make suggestions to our members in features like People You May Know. Unsubscribe here: https://www.linkedin.com/e/v2?e=-jrzliy-iob6c77x-m9&t=lun&midToken=AQFmrFVhGEd36Q&ek=second_guest_reminder_01&li=15&m=unsub&ts=HTML&eid=-jrzliy-iob6c77x-m9&loid=AQEtcXuQmx43wgAAAVS-0DYASddMtZsFZ1ZEfmqIUGAfjvGsOqqCGcPldL9iEizYobO6uj5pvevxnWqPVYfP17GKExtA5ZqODWFHuQ5YM636Mravv_Y8iZL5xR-rmmVhyztLLfTq1eOTepwuu8WMO03aCv_3XeVB5aDp36-BX4gAZaC9aQ8NrNyVDG_ycjLNE5UA

This email was sent to reply@reply.github.com.

If you need assistance or have questions, please contact LinkedIn Customer Service: https://www.linkedin.com/e/v2?e=-jrzliy-iob6c77x-m9&a=customerServiceUrl&ek=second_guest_reminder_01

&copy; 2016 LinkedIn Corporation, 2029 Stierlin Court, Mountain View CA 94043. LinkedIn and the LinkedIn logo are registered trademarks of LinkedIn.