mavlink/qgroundcontrol #1912: Firmware upgrade sync error: "Invalid sync response: 0x21 0x20"

bradsedito, 2015-10-03:
I'm having an issue trying to upgrade the firmware on my Pixhawk..  I keep getting the following when trying to upgrade my board: 

"Error: Sync: Send Command: Invalid sync response: 0x21 0x20"

I saw there was a posted solution to a previous post about this same problem, but I can't seem to get any of said solutions to work. Any ideas? I'm losing sleep over this one.

LorenzMeier, 2015-10-03:
You are on Linux, yes? Deinstall the modem manager in that case.

bradsedito, 2015-10-03:
Mac OSX El Capitan

bradsedito, 2015-10-03:
Should have probably mentioned that first. Haha. Sorry about that. How do I go about nixing the modem manager? Terminal maybe?

DonLakeFlyer, 2015-10-04:
modemmanager is not needed on OSX. To help debug, run the latest daily build with the command line option —logging:FirmwareUpgradeLog. Capture the console output from a failed flash and send it to us.

LorenzMeier, 2015-10-07:
Verified on Don's El Capitan install.

DonLakeFlyer, 2015-10-07:
Firmware Upgrade is working fine for me on El Capitan. It's regular connect that is flaky. 75% of the time it just hangs QGC. @bradsedito Need to log output to debug this.

DonLakeFlyer, 2015-10-09:
@bradsedito Can you provide a log as specified above?

bradsedito, 2015-10-12:
Sorry for the delay guys, birthday weekend xD how do I obtain the log in QGC?

DonLakeFlyer, 2015-10-12:
> To help debug, run the latest daily build with the command line option —logging:FirmwareUpgradeLog. Capture the console output from a failed flash and send it to us.

bradsedito, 2015-10-16:
How exactly do I do that? My apologies, having a hard time figuring it out

DonLakeFlyer, 2015-10-17:
What operating system are you using?

bradsedito, 2015-10-19:
Mac OSX El Capitan, using built-in terminal

DonLakeFlyer, 2015-10-19:
In terminal:
```
./Applications/QGroundControl.app/Contents/MacOS/qgroundground --logging:FirmwareUpgradeLog
```

julianoes, 2019-08-27:
I don't see this error anymore in the latest daily.

closed 2019-08-27 by julianoes