mavlink/qgroundcontrol #1914: Unable to locate package qt54qbs

zshn25, 2015-10-05:
I tried building QGCS on Ubuntu 14.04.3. I followed the instructions but I am getting this error when I run
```
sudo apt-get update && sudo apt-get install qt54tools qt54base qt54declarative qt54serialport qt54svg qt54webkit qt54quickcontrols qt54xmlpatterns qt54x11extras qt54websockets qt54sensors qt54script qt54quick1 qt54qbs qt54multimedia qt54location qt54imageformats qt54graphicaleffects qt54creator qt54connectivity libsdl1.2-dev libudev-dev
```
I get the following error
```
Reading package lists... Done                                                  
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package qt54creator is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package qt54qbs
E: Package 'qt54creator' has no installation candidate

```
[image]

I did System Settings:Software & Updates:Other Software and edit the entry for opt-qt541-trusty to reference Distribution: trusty.
I did
```
sudo apt-get clean
sudo apt-get update
sudo apt-get install
sudo apt-get install build-essential
sudo apt-get upgrade
```
and all these commands are executing without any error.

dagar, 2015-10-05:
See #1825

We need to update the instructions. For now you can remove qt54creator and qt54qbs and still have everything you need to build.

    sudo apt-get install qt54tools qt54base qt54declarative qt54serialport qt54svg qt54webkit qt54quickcontrols qt54xmlpatterns qt54x11extras qt54websockets qt54sensors qt54script qt54quick1  qt54multimedia qt54location qt54imageformats qt54graphicaleffects qt54connectivity libsdl1.2-dev libudev-dev

DonLakeFlyer, 2015-10-05:
Try this and let me know if it works:
- I don’t think you need qt54qbs, just skip it
- Use the QtCreator that comes pre-installed with Ubuntu

zshn25, 2015-10-06:
Thanks @dagar, it worked.

closed 2015-10-06 by DonLakeFlyer