mavlink/qgroundcontrol #1921: Joystick code should support mavlink generic vehicles

DonLakeFlyer, 2015-10-07:


v01d, 2016-10-17:
I'm developing a custom firmware for a ground rover vehicle and I was expecting virtual joystick of Android version to work when, after looking at the code, this is simply disabled for generic autopilots. I don't see why this shouldn't be disabled by default since it would simply mean the message is ignored if it is not supported.

Alternatively, the MAV_PROTOCOL_CAPABILITY enum could be extended to know about the autopilot's capability.

DonLakeFlyer, 2016-10-17:
Can you point me at the code that disables it?

v01d, 2016-10-17:
The line in question is this:
https://github.com/mavlink/qgroundcontrol/blob/master/src/FirmwarePlugin/FirmwarePlugin.cc#L92
which should be true as with PX4

https://github.com/mavlink/qgroundcontrol/blob/master/src/FirmwarePlugin/PX4/PX4FirmwarePlugin.cc#L173

I understand this change should be enough since the check is as follows:
https://github.com/mavlink/qgroundcontrol/blob/cbee0fb9aab043924b4c5df7bbc3d62685522ff1/src/FlightDisplay/FlightDisplayView.qml#L94

DonLakeFlyer, 2016-10-17:
Thanks I couldn't find it. Yes we could back off on turning if off and just leave the warning in.

v01d, 2016-10-17:
Ok!

Are there bleeding edge android builds to test this once it is in master?

DonLakeFlyer, 2016-10-17:
Yes, there are. But there were some problems there.

@dagar What's ths status of Android Daily/Stable?

dagar, 2016-10-17:
Back to normal. Android Stable is v3.0.1 and Beta is 42f447c from yesterday.

v01d, 2016-10-17:
How can I try the nightly android builds?

dagar, 2016-10-17:
Either opt in to the Beta through the Play store listing, or sideload this - https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/QGroundControl.apk

v01d, 2016-10-18:
Ok, let me know then when the change is made and I can test it.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier