mavlink/qgroundcontrol #1925: Master QGC doesn't upgrade firmware

FPVflyer, 2015-10-07:
Start bootload, then broke connection. 
[image]
see screenshot.

LorenzMeier, 2015-10-07:
Is this Mac OS X El Capitan?

FPVflyer, 2015-10-07:
Yes

Date: Wed, 7 Oct 2015 13:08:11 -0700

DonLakeFlyer, 2015-10-07:
Can you try a firmware upgrade with --logging:FirmwareUpgradeLog on the command line. Capture the console output and post it here.

Those wacky chars in the setup page titles is an El Capitan thing as well. Great OS X release so far! Screwed up a pile of stuff. Thanks Apple (or maybe Qt).

DonLakeFlyer, 2015-10-13:
> Can you try a firmware upgrade with --logging:FirmwareUpgradeLog on the command line. Capture the console output and post it here.

Hello?

FPVflyer, 2015-10-13:
Hi Don,
Sorry have been very busy, I'll
Probably try tonight.

Sent with my phone

closed 2015-12-03 by DonLakeFlyer