mavlink/qgroundcontrol #1928: MSVC parallel build is quadratic using QtCreator/jom

NaterGator, 2015-10-08:
Since the jom builder already spawns n processes for a n-cored computer, specifying the -MP flag [here](https://github.com/mavlink/qgroundcontrol/blob/2cf5a34eb13819c18e194846c6f46175f5fda810/QGCCommon.pri#L171) actually causes n<sup>2</sup> parallel compilation threads. 

I'd submit a patch to change this default behavior but I'm not sure if anyone is still building with MSVC or using the nmake builder on windows instead of jom. My build times got substantially better without the -MP flag.

DonLakeFlyer, 2015-10-08:
This is a Qt problem not QGC. We don't have control over that level of the makefiles that are created by qmake. How are you going to patch it?

NaterGator, 2015-10-08:
I would argue if the [recommended](http://qgroundcontrol.org/dev/start#an_important_note) IDE is QtCreator that we remove the -MP flag from the common pri. Windows build automation should explicitly use jom instead of nmake; the infrequent Visual Studio IDE developer would need to beware to use jom or add the -MP flag explicitly. 

Jom has a few advantages over -MP including parallel UI / MOC generation and more reliable pdb generation. It is strictly a Windows developer quality-of-life change, but I have always found jom to be a faster and more efficient means of parallel compilation using Microsoft's compilers.

DonLakeFlyer, 2015-10-08:
The recommended Windows IDE is VS since it requires no additional installs.

LorenzMeier, 2015-10-08:
Don, are you still running the qmake step on the console? Try using Qt Creator in that case. I would expect it will be a much smoother experience, in particular for Windows devs who are not accustomed to running stuff in the console.

DonLakeFlyer, 2015-10-08:
Sure, but native debugging in VS is way better. Qt Creator is fine for someone who just builds and never gets stuck having to debug anything.

NaterGator, 2015-10-08:
@DonLakeFlyer that's a separate issue, but I'd argue exactly the opposite. QtCreator's CDB plugin provides for much prettier / logical debugging of native Qt types without significantly hampering the rest of the debugging experience on Windows. It lacks a great deal in profiling support, but again that is for another discussion. 

I submitted a pull request which provides a message during qmake invocation to warn developers who may be using jom to pass CONFIG+=jombuild to qmake. I also added the config to the travis build script so we can compare the build time before/after. 

At this stage it is entirely opt-in behavior that will improve quality of life for anyone hoping to use QtCreator on Windows. For Windows users in creator 3.5.0 the setting looks like this:
[image]
under the "Projects" tab.

Hopefully this is a net improvement and doesn't intrude. Just want to make things a little easier for those of us stuck in Windows :)

DonLakeFlyer, 2015-10-08:
Ok, I'll have to try it again. Once AppVeyor is fully online for Windows builds and I can turn off TC for Windows we can change this for real. For now, I don't want to have to update TC to a new model. When that happens can I ping you to update readme and build files to point in the right direction?

NaterGator, 2015-10-08:
I think this solution is workable for now and should help speed up the AppVeyor builds. It may also be worthwhile to let jom automatically manage the level of parallelism and omit the -j4 argument to jom in the appveyor script. For now though I think this is good enough to call fixed. Thanks!

closed 2015-10-08 by NaterGator