mavlink/qgroundcontrol #1934: Unneeded extra file compilation with specific hierarchy of build and project folders

sveta-soltysyak, 2015-10-09:
I've rebuild the project which took approx 15 minutes, then I've changed a single file **qgroundcontrol\src\ui\map\QGCMapToolBar.cc** and ran build (not rebuild) but **seems like a rebuild was performed** - it took approx the same 15 minutes. The issue appeared when I was using QtCreator. When I was using Visual Studio problem did not appear - the same file change caused 15 sec build (rebuild in VS is 15 minutes).

After some investigations, turned out that **problem is reproducible with specific build and project folders location**. 
My project is located in folder **'<Root>\QGC\qgroundcontrol'**. In QtCreator BuildSettings **BuildDirectory** is set to **'<Root>\QGC\qgcbuilddirectory'**. After running qmake the makefiles are generated into this '<Root>\QGC\qgcbuilddirectory' folder. Looking into **Makefile.QGCApplication.Debug** I see that at **build-debug\ui\ui_QGCQmlWidgetHolder.h:** there is dependency on **..\qgroundcontrol\src\ui\map**  (the folder where file QGCMapToolBar.cc is located) instead of **C:\Qt\Qt5.4.0\5.4\msvc2013_opengl\include\QtCore\qmap.h** (as in the correctly generated makefile when BuildDirectory is same as project directory '<Root>\QGC\qgroundcontrol'). And I see more lines like this in the makefile.

Could this be a bug in qmake, or in .pro files? Thank you.

DonLakeFlyer, 2015-10-13:
We don't have very much control at this level of how qmake creates makefiles. If you build from the command line using qmake -r qgroundcontrol.pro, things work correctly. Also if you use the QtCreator defaults things work correctly. I'd suggest using on of the options.

closed 2015-10-13 by DonLakeFlyer

Vavooon, 2015-10-13:
Unfortunately it doesn't work for us with default Qt Creator settings, but it works great with disabled Shadow Build feature.