mavlink/qgroundcontrol #1938: Mode switch assignment changes are only applied when switching away from the window

LorenzMeier, 2015-10-10:
@DonLakeFlyer The mode switch config UI applies the changes you do once you switch away from the window, which is very confusing if you set it up and then try to arm and test. Given that the GUI has a low change / click count, how about applying the change immediately whenever you change an option, or alternatively clearly indicating with a right-menu that you need to save these?

DonLakeFlyer, 2015-10-10:
> The mode switch config UI

Not sure what you are talking about. Changing flight modes from the toolbar? Setting up Flight Modes using the Setup/Flight Modes?

LorenzMeier, 2015-10-10:
The flight mode config UI in the setup tab.

DonLakeFlyer, 2015-10-10:
Do you have an example case? That isn't the way flight mode config works. Changes are applied immediately, there is no specific save step.

LorenzMeier, 2015-10-10:
Are you putting the system into the RC cal mode while you show the UI? That would disable onboard mode switch handling. Does it switch modes for you in the top status bar?

DonLakeFlyer, 2015-10-10:
> Are you putting the system into the RC cal mode

Nope.

> Does it switch modes for you in the top status bar?

I'll give it a try later.

DonLakeFlyer, 2015-10-10:
> Are you putting the system into the RC cal mode

Although now that I think about it, that's probably not a good idea? Probably should go into RC Cal mode don't you think?

LorenzMeier, 2015-10-10:
No, you shouldn't. If the system is disarmed exercising the mode switches is fine, and you block it when its armed.

RC cal != mode switch setup

DonLakeFlyer, 2015-10-10:
I just tried it and it works fine. The toolbar showed mode changes as I was adjusting flight mode config. I did it with a real RC connection.

LorenzMeier, 2015-10-12:
I can't repro either - I must have been mistaken.

closed 2015-10-12 by LorenzMeier