mavlink/qgroundcontrol #1950: Segfault when trying to enter Joystick tab

tumbili, 2015-10-11:
I tried connecting a Sony and a Speedlink Joystick to QGC. I pulled and built the latest master.
When I clicked on the Joystick tab QGC closed due to a segmentation fault. I will see tomorrow in the lab if I get the same behaviour with my Logitech controller which I used before.
The controller I used today appeared as /dev/input/js1 and I successfully read data using "cat" or "jstest".

DonLakeFlyer, 2015-10-11:
I can't repro with my Ubuntu and Logitech F710. If possible someone will need to get this in the debugger.

LorenzMeier, 2015-10-11:
@tumbili Try a fresh build (delete the build folder) and then check what the calibration values look like at this point:
https://github.com/mavlink/qgroundcontrol/blob/master/src/VehicleSetup/JoystickConfigController.cc#L518

Most likely they're intact here and then get trashed after.

@devbharat Either you or @tumbili will need to debug and fix this - we've maxed out all remote debugging options, including debugging on an Ubuntu VM with the Logitech and with the Taranis.

LorenzMeier, 2015-10-11:
Closing this in favour of #1850.

closed 2015-10-11 by LorenzMeier