mavlink/qgroundcontrol #1955: Mission check: False positives?

LorenzMeier, 2015-10-12:
@DonLakeFlyer I (and Roman) get this at times:
```
Error during Mission communication with Veicle: Vehicle responded incorreclty to mission item protocol sequence: No Ack: MISSION_COUNT
```
I suspect that this is due to the mission request being timed on the link connect, rather than the first heartbeat. If it is a true positive let us know and we'll debug the onboard implementation.

DonLakeFlyer, 2015-10-12:
Everything happens after the first hearbeat from the Vehicle. Mission request will be sent after that, wait for 2 seconds for an ack to come back. If that fails it will retry 5 times.

That said, this looks like something wrong on the QGC side. The "No Ack" part of the message signals something set up incorrectly. Will need a run with MissionManagerLog to figure out what is wrong.

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier