mavlink/qgroundcontrol #1958: Arm and flight mode toolbar plugins stop working in HIL

AndreasAntener, 2015-10-12:
On latest master: the new toolbar plugins for arming/disarming and flight-mode switching stop working as soon as I'm connected to the simulator (X-Plane 10 in my case). Before I'm connected to the simulator the commands get through. I debugged while connected and I see the send get called. Maybe it's a buffer issue?

It still worked for me with the old command widget, although I didn't pull for a while so the UI change is probably unrelated.

DonLakeFlyer, 2015-10-12:
Let me give it a try.

DonLakeFlyer, 2015-10-12:
I tested:
- Master QGC + Master firmware
- Airframe: HIL quadcopter X
- jmavsim: Arm/Disarm working
- FlightGear (using widget): Arm/Disarm working

AndreasAntener, 2015-10-12:
Tried too with FlightGear now, no luck..

DonLakeFlyer, 2015-10-12:
Do you get any error from the firmware telling you that it couldn't arm? In the toolbar message dropdown.

AndreasAntener, 2015-10-12:
Nope. I get the correct feedback though when I'm not connected to the sim, e.g. command unsupported

DonLakeFlyer, 2015-10-12:
> e.g. command unsupported

Huh? Where is that coming from?

AndreasAntener, 2015-10-12:
Sorry I mean "temporarily rejected", playing too much with commands right now ;)

DonLakeFlyer, 2015-10-12:
My guess is that the firmware is rejecting the arm for some reason which is related to you vehicle HIL setup. Probably need to debug from the firware side.

AndreasAntener, 2015-10-12:
This message I only get when I'm not connected to hil, which makes sense. Just out of curiosity I routed the mavlink connection via a UDP proxy (between qgc and usbserial), then it works "sometimes". I'm still suspecting that some messages are getting dropped.

AndreasAntener, 2015-10-12:
I'll try debugging it. Did the usb/serial handling change between now and 2.7.1?

DonLakeFlyer, 2015-10-12:
Don't think so

LorenzMeier, 2015-10-12:
@AndreasAntener Are you on El Capitan? If yes, then Apple screwed you.

DonLakeFlyer, 2015-10-12:
> I tested:
> 
> Master QGC + Master firmware
> Airframe: HIL quadcopter X
> jmavsim: Arm/Disarm working
> FlightGear (using widget): Arm/Disarm working

This was all tested on El Capitan. Working fine. On El Capitan it will just fail to connect over usb about 10% of the time.

AndreasAntener, 2015-10-13:
I tried debugging this yesterday but had no luck. I'm not on El Capitan yet, I usually wait a few months before I update something from Apple ;)

Also I downloaded the daily QGC build to rule out build issues, same thing really. Sometimes the message gets through, maybe in 1 out of 20. As I said, it works with 2.7.1., so something must have changed. Maybe the simulator interface? More throughput?

DonLakeFlyer, 2015-10-13:
I think maybe in 2.7.1 is was using MAV_CMD_DO_SET_MODE to arm/disarm whereas now it used MAV_CMD_COMPONENT_ARM_DISARM. But set flight mode would be the same command. HIL simulator code hasn' t been touched in ages.

AndreasAntener, 2015-11-01:
I just tried all the latest code again, and still have the same problem. The behaviour changed a bit though: when QGC is not connected to HIL, the commands go through (and get rejected). When QGC is connected to HIL, I get the "ekf stale sensor data" message right after every command I issue (arm/disarm and mode switch). Does this tell anyone something?

LorenzMeier, 2015-11-01:
Let me debug this, I can reproduce it now.

LorenzMeier, 2015-11-01:
FYI: Also fixing this right now: https://github.com/PX4/Firmware/issues/3107

Its flyable but odd.

LorenzMeier, 2015-11-01:
Seems to be on the QGC side. The command never makes it to the autopilot.

DonLakeFlyer, 2015-11-02:
I won't have time to look at this before I leave. But if you can repro and have time you can start debugging at Vehicle::setFlightMode.

DonLakeFlyer, 2015-12-03:
Still a problem?

AndreasAntener, 2015-12-04:
I think Lorenz was able to resolve this on the Firmware side.

closed 2015-12-04 by AndreasAntener