mavlink/qgroundcontrol #1959: New simulation pane does not fully delete HIL control link when removed

LorenzMeier, 2015-10-12:
@DonLakeFlyer The new HIL control widget logic does not properly destroy the simulation link, requiring a QGC restart when you reboot the autopilot. @AndreasAntener will fix this, but for now just keep restarting QGC when you reconnect.

DonLakeFlyer, 2015-10-12:
When you disconnect, the Vehicle should go away which should in turn stop the hil link:

```
UAS::~UAS()
{
#ifndef __mobile__
    stopHil();
    if (simulation) {
        // wait for the simulator to exit
        simulation->wait();
        simulation->deleteLater();
    }
#endif
    writeSettings();
}
```

If this is not working, not sure it has anything to do with the new widget code. That didn't really change any logic behind it.

DonLakeFlyer, 2015-10-12:
Either way, just rebooting the vehicle would not have stopped the hil link. You would have had to disconnect. Rebooting the vehicle just results in a lost connection, which doesn't trigger anything.

DonLakeFlyer, 2015-10-12:
Tested Start HIL and then Disconnect from toolbar. FlightGear was left running. So something is wrong there. Let me look. Start/Stop on widget multiples times works ok.

DonLakeFlyer, 2015-10-12:
Looking at this more, FlightGear was never shut down in this case. The the link is stopped. So everything still seems the same to me. What are the exact repro steps that cause a problem.

I do see a possible problem on the firmware sid. When I do this:
- Connect
- Start from widget
- Wait for FlightGear to start up
- Stop from widget

I get an infinite stream of these from the firmware serial port:
WARN  53422229   timed out (file /home/travis/build/PX4/Firmware/src/modules/navigator/navigator_main.cpp line 340)

LorenzMeier, 2021-01-22:
Closing as stale.

closed 2021-01-22 by LorenzMeier