mavlink/qgroundcontrol #1962: MAVLINK_MSG_ID_REQUEST_DATA_STREAM missing?

krzysztofkuczek, 2015-10-12:
Hi,
Years ago i used qground control to comunicate with my own FC, and there was option to request data streams, to setup streams frequency etc.
How cat I force 2.7 version to send such a message?
Or maybe i can download 1.x version from somewhere?
Best regards
Krzysiek

DonLakeFlyer, 2015-10-12:
For a generic mavlink connection QGC now expects streams to be setup up automatically on the firmware side. There is no longer a UI to request them manually.

krzysztofkuczek, 2015-10-12:
OK, Thanks for explanation.
Any chance to download older version for windows?

DonLakeFlyer, 2015-10-12:
Don’t think anything we still have around has that in it.

DonLakeFlyer, 2015-10-12:
Closing, since already covered in: Issue #1941

closed 2015-10-12 by DonLakeFlyer